aws/aws-fpga #337: You are not licensed to encrypt IEEE1735 Version 2

TomKeddie, 2017-05-15:
@aws/f1-fpga-hardware-development-kit

Release version
HDK_VERSION=1.2.0

Expected Behavior
examples build as described

Actual Behavior
both cl_dram_dma and cl_hello_world fail with the above message on a new AMI image created today.

Steps to reproduce
```
[centos@ip-172-31-15-221 scripts]$  cd /home/centos/aws-fpga/hdk/cl/examples/
[centos@ip-172-31-15-221 cl_dram_dma]$ export CL_DIR=$(pwd)
[centos@ip-172-31-15-221 cl_dram_dma]$  cd $CL_DIR/build/scripts
[centos@ip-172-31-15-221 scripts]$ ./aws_build_dcp_from_cl.sh -notify

INFO: Starting the design checkpoint build process
INFO: Checking for proper environment variables and build directories
INFO: Environment variables and directories are present. Checking for Vivado installation.
INFO: Build through Vivado is running as background process, this may take few hours.
INFO: Output is being redirected to 17_05_15-183826.nohup.out
INFO: If you have set your EMAIL environment variable and -notify is specified, you will receive a notification when complete.
INFO:   (See $HDK_DIR/cl/examples/README.md for details)
[centos@ip-172-31-15-221 scripts]$ tail last_log
## file copy -force $UNUSED_TEMPLATES_DIR/unused_aurora_template.inc  $TARGET_DIR
## file copy -force $UNUSED_TEMPLATES_DIR/unused_hmc_template.inc     $TARGET_DIR
## file copy -force $UNUSED_TEMPLATES_DIR/unused_sh_bar1_template.inc $TARGET_DIR
## exec chmod +w {*}[glob $TARGET_DIR/*]
## encrypt -k $HDK_SHELL_DIR/build/scripts/vivado_keyfile.txt -lang verilog  [glob -nocomplain -- $TARGET_DIR/*.?v] [glob -nocomplain -- $TARGET_DIR/*.vh] [glob -nocomplain -- $TARGET_DIR/*.inc]
ERROR: [Vivado 12-3330] encrypt 1735: Failed to encrypt ''./home/centos/aws-fpga/hdk/cl/examples/cl_dram_dma/build/src_post_encryption/.temp/vivado_keyfile.txt: You are not licensed to encrypt IEEE1735 Version 2 [/home/centos/aws-fpga/hdk/cl/examples/cl_dram_dma/build/src_post_encryption/.temp/vivado_keyfile.txt]

    while executing
"source encrypt.tcl"
    (file "create_dcp_from_cl.tcl" line 167)

```

sbarillet, 2017-05-16:
Did you use amazon instance( and their installed tools) in the instance  to synthesize the design or you are working to a local machine ?
If the second case, you need to have the on-premise xilinx licenses installed. The licenses provide a "EncryptedWriter_v2" feature required to perform P1735 v2.0 encryption

TomKeddie, 2017-05-16:
> Did you use the amazon FPGA Developper AMI instance (and their installed tools) to synthesize the example design

Thanks for your suggestions, I'm using an ec2 instance created from the aws fpga AMI yesterday.  I did also get a standalone license from xilinx but was unable to add it to the ec2 instance.  Either I missed a step or the AMI has an issue.

kristopk, 2017-05-16:
If you have launched your instance on AWS EC2 using the FPGA AMI, Xilinx licenses are built-in.  Thanks!

TomKeddie, 2017-05-16:
> If you have launched your instance on AWS EC2 using the FPGA AMI, Xilinx licenses are built-in. Thanks!

Yes, I would expect so but I can't build the example designs.

kristopk, 2017-05-25:
Assuming the issue has been resolved per the thread posted on the forum:
Hi Tom,
I sent you a PM on 5/20. 
I suspect that with multiple users you are hitting a known issue with the AMI. Our AMI has the license in:
$cd /home/centos/src/license
$ls
XilinxAWS.lic
We have realized this was a bad place to keep the license and fails to load when multiple user accounts are created. 
Another thing to check is:
$ printenv | grep license
XILINXD_LICENSE_FILE=/home/centos/src/license/XilinxAWS.lic

Please change the location of the license and set env vars as needed for your setup. 

Our next AMI release will fix this issue.

Thanks
Kris

closed 2017-05-25 by kristopk

ShadowTSD, 2020-09-06:
你好，能否分享一下你的IEEE1735 v2 encryption license，谢谢了

ShadowTSD, 2020-09-06:
Hello, can you share your ieee1735 V2 encryption license? Thank you

deeppat, 2020-09-07:
@ShadowTSD You should be able to checkout a license from the FPGA Developer AMI. Are you having issues with that?

ShadowTSD, 2020-09-07:
xilinx solution given on the website is: send an email application, but I have not received a reply after sending an email. Do you have this license? Can you share it, thank you.

| |
唐山东
|
|
邮箱：tangshandd@163.com
|

签名由 网易邮箱大师 定制

On 09/08/2020 00:25, Deep Patel wrote:

@ShadowTSD You should be able to checkout a license from the FPGA Developer AMI. Are you having issues with that?

deeppat, 2020-09-08:
@ShadowTSD: Our FPGA Developer AMI launched through the AWS Marketplace should provide this license for you. Xilinx tools should provide IEEE 1735 V2 encryption when you use the FPGA Developer AMI. 
Since we do not provide or share license files anymore. I think your best bet to get a license file is from Xilinx directly.

ShadowTSD, 2020-09-09:
Yes, I sent an email to Xilinx, but they didn't reply. Thank you for your reply.