aws/aws-fpga #356: Typo in hdk/common/verif/include/sh_dpi_tasks.svh

acanis, 2017-09-05:
There is a typo in the peek_bar1 and peek_sda tasks. The peek_ocl is correct. See patch below:

@@ -430,7 +440,7 @@ end
              output logic [63:0] data,
              input logic [5:0] id = 6'h0);
        logic [63:0] tmp;
-       `SLOT_MACRO_TASK(peek(.addr(addr), .data(data), .id(id), .size(DataSize::UINT32), .intf(AxiPort::PORT_SDA)))
+       `SLOT_MACRO_TASK(peek(.addr(addr), .data(tmp), .id(id), .size(DataSize::UINT32), .intf(AxiPort::PORT_SDA)))
        data = tmp[31:0];
    endtask

@@ -468,7 +478,7 @@ end
              output logic [63:0] data,
              input logic [5:0] id = 6'h0);
        logic [63:0] tmp;
-       `SLOT_MACRO_TASK(peek(.addr(addr), .data(data), .id(id), .size(DataSize::UINT32), .intf(AxiPort::PORT_BAR1)))
+       `SLOT_MACRO_TASK(peek(.addr(addr), .data(tmp), .id(id), .size(DataSize::UINT32), .intf(AxiPort::PORT_BAR1)))
        data = tmp[31:0];
    endtask

AWShimasajj, 2017-09-06:
Thanks for letting us know about this issue. We will make a fix and upload the patch soon.

kristopk, 2017-11-21:
released as part of v1.3.4

closed 2017-11-21 by kristopk