aws/aws-fpga #393: How can I get the reports after running on F1

AlverLyu, 2018-03-24:
After running the hardware emulation of the Xilinx example helloworld_ocl with  $AWS_PLARTFORM_4DDR_DEBUG, it would generates the profile reports. While after running in the FPGA, there's no reports generated. I have tried to modify the makefile `rules.mk` to add a `--debug` option for `xocc`, which does not help. How can I get these reports for the hardware running?

ThomasXilinx, 2018-03-29:
By default, reporting is:
- Enabled for the emulation flows
- Disabled when running on the FPGA

To enable reporting when running on the FPGA, create a file called sdaccel.ini in the run directory with the following content:

> [Debug]
>  timeline_trace=true
>  profile=true

The content of the sdaccel.ini file control which reports get generated during both emulation and FPGA runs.

AlverLyu, 2018-03-30:
@ThomasXilinx This exactly solved my problem. Thank you very much.

closed 2018-03-30 by AlverLyu