aws/aws-fpga #400: sdaccel_setup.sh fails on aws fpga dev ami

paulbellamy, 2018-05-13:
In the Ireland region, after starting a c4.4xlarge with the ami `FPGA Developer AMI - 1.4.0 - pre8-40257ab5-6688-4c95-97d1-e251a40fd1fc-ami-0335b86e84e820e8d.4 (ami-c6be8bbf)`, when running `sdaccel_setup.sh`, I get an error like:

```
install: cannot stat ‘/opt/Xilinx/SDx/2017.4.op/lib/lnx64.o/libstdc++.so*’: No such file or directory
make: *** [install] Error 1
ERROR: Install of SDAccel runtime FAILED
```

In the eu-west-1 region, with ami `FPGA Developer AMI - 1.4.0 - pre8-40257ab5-6688-4c95-97d1-e251a40fd1fc-ami-0335b86e84e820e8d.4 (ami-f7ba3988)`, it gives the same error.

Maybe there is a new version of the AMI, and the script needs updating?

Full log:
```
[centos@ip-172-30-1-66 ~]$ git clone https://github.com/aws/aws-fpga.git $AWS_FPGA_REPO_DIR
Cloning into '/home/centos/src/project_data/aws-fpga'...
remote: Counting objects: 5550, done.
remote: Compressing objects: 100% (362/362), done.
remote: Total 5550 (delta 131), reused 324 (delta 94), pack-reused 5065
Receiving objects: 100% (5550/5550), 224.20 MiB | 27.06 MiB/s, done.
Resolving deltas: 100% (2779/2779), done.
Checking out files: 100% (1201/1201), done.
[centos@ip-172-30-1-66 ~]$ cd $AWS_FPGA_REPO_DIR
[centos@ip-172-30-1-66 aws-fpga]$ source sdaccel_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR=/home/centos/src/project_data/aws-fpga
INFO: Sourcing sdk_setup.sh
INFO: Setting up environment variables
DEBUG: AWS_FPGA_REPO_DIR=/home/centos/src/project_data/aws-fpga
Entering /home/centos/src/project_data/aws-fpga/sdk/userspace/utils
rm -f *.o *.a ../lib/libutils.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o io.o io.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o log.o log.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o logger-kmsg.o logger-kmsg.c
mkdir -p ../lib
ar rcs ../lib/libutils.a io.o log.o logger-kmsg.o
Entering /home/centos/src/project_data/aws-fpga/sdk/userspace/fpga_libs/fpga_pci
rm -f *.o ../../lib/libfpga_pci.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_pci.o fpga_pci.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_pci_sysfs.o fpga_pci_sysfs.c
mkdir -p ../../lib
ar rcs ../../lib/libfpga_pci.a fpga_pci.o fpga_pci_sysfs.o
Entering /home/centos/src/project_data/aws-fpga/sdk/userspace/fpga_libs/fpga_mgmt
rm -f *.o ../../lib/libfpga_mgmt.a ../../lib/so/libfpga_mgmt.so.1.0.0
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_mgmt.o fpga_mgmt.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_mgmt_cmd.o fpga_mgmt_cmd.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_hal_mbox.o fpga_hal_mbox.c
mkdir -p ../../lib
for library in ../../lib/libfpga_pci.a ../../lib/libutils.a; do ar -x $library; done
ar rcs ../../lib/libfpga_mgmt.a *.o
mkdir -p ../../lib/so
cc -o ../../lib/so/libfpga_mgmt.so.1.0.0 -shared fpga_mgmt.o fpga_mgmt_cmd.o fpga_hal_mbox.o -Wl,-soname,libfpga_mgmt.so.1 -Wl,--whole-archive ../../lib/libfpga_pci.a ../../lib/libutils.a -Wl,--no-whole-archive -lrt -lpthread
ln -sf libfpga_mgmt.so.1.0.0 ../../lib/so/libfpga_mgmt.so
Entering /home/centos/src/project_data/aws-fpga/sdk/userspace/fpga_mgmt_tools/src
rm -f *.o *.a fpga-local-cmd static-fpga-local-cmd
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o fpga_local_cmd.o fpga_local_cmd.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o fpga_local_cmd_parse.o fpga_local_cmd_parse.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o virtual_jtag_pcie.o virtual_jtag_pcie.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o virtual_jtag_server.o virtual_jtag_server.c
cc -o fpga-local-cmd fpga_local_cmd.o fpga_local_cmd_parse.o virtual_jtag_pcie.o virtual_jtag_server.o -L../../lib/so -lfpga_mgmt
cc -o static-fpga-local-cmd fpga_local_cmd.o fpga_local_cmd_parse.o virtual_jtag_pcie.o virtual_jtag_server.o -L../../lib -lfpga_mgmt -lrt -lpthread
Build complete.

Root privileges are required to install. You may be asked for your password...
Executing as root...

AWS FPGA: Copying Amazon FPGA Image (AFI) Management Tools to /usr/bin
AWS FPGA: Installing shared library to /usr/local/lib64
	libfpga_mgmt.so.1 (libc6,x86-64) => /usr/local/lib64/libfpga_mgmt.so.1
AWS FPGA: Done with Amazon FPGA Image (AFI) Management Tools install.
[... 101 lines cut ...]
make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build M=/home/centos/src/project_data/aws-fpga/sdk/linux_kernel_drivers/xdma modules_install
make[2]: Entering directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
  INSTALL /home/centos/src/project_data/aws-fpga/sdk/linux_kernel_drivers/xdma/xdma.ko
Can't read private key
  DEPMOD  3.10.0-693.21.1.el7.x86_64
make[2]: Leaving directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
depmod -a
install -m 644 10-xdma.rules /etc/udev/rules.d
rmmod -s xdma || true
modprobe xdma
make[1]: Leaving directory `/home/centos/src/project_data/aws-fpga/sdk/linux_kernel_drivers/xdma'
install -d /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/platforms/xilinx_aws-vu9p-f1_1ddr-xpr-2pr_4_0/driver
install -d /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/bin
install -d /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/lib/x86_64
install -m 755 /home/centos/src/project_data/aws-fpga/SDAccel/userspace/src/libawsxcldrv.so /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/platforms/xilinx_aws-vu9p-f1_1ddr-xpr-2pr_4_0/driver
install -m 755 /home/centos/src/project_data/aws-fpga/SDAccel/tools/awssak/xbsak /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/bin
install -m 755 /opt/Xilinx/SDx/2017.4.op/runtime/bin/xclbincat /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/bin
install -m 755 /opt/Xilinx/SDx/2017.4.op/runtime/bin/xclbinsplit /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/bin
#install -m 755 /opt/Xilinx/SDx/2017.4.op/lib/lnx64.o/libxilinxopencl.so /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/lib/x86_64
install -m 755 /home/centos/src/project_data/aws-fpga/SDAccel/aws_platform/libxilinxopencl.so /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/lib/x86_64
install -m 755 /opt/Xilinx/SDx/2017.4.op/lib/lnx64.o/libstdc++.so* /opt/Xilinx/SDx/2017.1.rte.1ddr/runtime/lib/x86_64
install: cannot stat ‘/opt/Xilinx/SDx/2017.4.op/lib/lnx64.o/libstdc++.so*’: No such file or directory
make: *** [install] Error 1
ERROR: Install of SDAccel runtime FAILED
```

kristopk, 2018-05-13:
Hi 

Below is a compatibility table describing the mapping of developer kit version to AMI version:  

| Developer Kit Version   | Tool Version Supported     |  Compatible AMI Version     |
|-----------|-----------|------|
| 1.3.0-1.3.6 | 2017.1 | v1.3.5 |
| 1.3.7 or greater | 2017.1 | v1.3.5 or greater (Xilinx SDx 2017.4) |
| 1.3.7 or greater | 2017.4 | v1.4.0 or greater (Xilinx SDx 2017.4) |

The current aws-fpga is version v1.3.6, therefore, please use AMI version v1.3.5.

Sorry for the confusion; we are in the process of releasing a new developer kit (v1.3.7).  More info can be found here:
https://forums.aws.amazon.com/thread.jspa?messageID=848073&#848073

Thanks
Kris

paulbellamy, 2018-05-13:
Ah, thanks. It might be nice to mention that in the Getting Started readmes, https://github.com/aws/aws-fpga/blob/master/SDAccel/README.md, so other people don't get confused. :)

kristopk, 2018-05-14:
Thanks for the feedback.  This info has been added to the SDAccel/README.md

closed 2018-05-14 by kristopk