aws/aws-fpga #405: SDAccel - helloworld_ocl

vegaluisjose, 2018-05-24:
I am getting some errors when I try to program/run the helloworld_ocl example on the recent upgrade. I am using AMI 1.4.0 and v1.3.7 (aws-repo). It seems a driver (userspace) issue, any thoughts?

`
Device/Slot[0] (/dev/xdma0, 0:0:1d.0)
xclProbe found 1 FPGA slots with XDMA driver running
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
Device/Slot[0] (/dev/xdma0, 0:0:1d.0)
xclProbe found 1 FPGA slots with XDMA driver running
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
Device/Slot[0] (/dev/xdma0, 0:0:1d.0)
xclProbe found 1 FPGA slots with XDMA driver running
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
opendir: Path /sys/bus/pci/devices/0000:00:1d.0/drm does not exist or could not be read: No such file or directory
[0]user:0x1042:0x7:[xdma:2017.1.47:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
Error: Failed to find Xilinx platform
`

AWScsaralay, 2018-05-24:
Hi,

Can you share the linux kernel version that you are using?

Thanks,
Chakra

vegaluisjose, 2018-05-24:
Hi,

I was using the kernel included in the FPGA developer AMI v1.4.0, I just
terminated the instance because it was not working.

By the way, do you happen to know the status of the datastream_pipes_ocl
example? and which combination of AMI and aws-fpga will make it work?

I saw that you guys are using xcl(old) on that example rather than xcl2
(new). I am trying to use OpenCL pipes for a project.

-
Luis Vega

kristopk, 2018-05-25:
For OpenCl pipes and xcl2 question, please create a thread on the AWS forum:
https://forums.aws.amazon.com/forum.jspa?forumID=243&start=0

For the driver issue, please send your commands you used prior to hitting this issue?  We are not seeing the issue you described in our testing and the team was not able to reproduce the error.  

btw: Your post shows the XDMA driver is running.  v1.4.0 github SDAccel setup does not use the XDMA driver, the setup should have loaded the XOCL driver.  
Kris

vegaluisjose, 2018-05-26:
Yeah, I think the main takeaway is don't "source sdaccel_setup.sh" from different aws-fpga versions on the same instance, otherwise both drivers will start causing issues. I will close the issue. Thanks!

closed 2018-05-26 by vegaluisjose