aws/aws-fpga #409: Unable to run helloworld_ocl on AMI 1.4.0

lyager, 2018-06-12:
On the latest AMI1.4.0 I tried compiling the helloworld_ocl example and run it on hardware. After uploading it my S3 bucket and programming it back to the FPGA I get an error on the first attempt, where the host program tries to create a CL::Buffer.

Adding error handling to the CL::Buffer it's clear that I get error '-6', which is CL_OUT_OF_HOST_MEMORY.

I tried modifying the helloworld example to just allocate 1 byte, result is the same. Can someone reproduce this?

-- Yes, I know I should have used to AWS forum, but apparently: 'Your message quota has been reached. Please try again later.' I posted 1 reply today... *sigh*

nikever, 2018-06-23:
Same thing here!

This is ridicolous, keep wasting money and time. This is a terrible release. 

As @lyager I can't open a thread on the AWS Forum because I've replied to a message. Message Quota to 1, really?

lyager, 2018-06-25:
@nikever just wanted to point you to a solution that we've figured out works: https://forums.aws.amazon.com/thread.jspa?threadID=283806&tstart=0

The trick for AMI1.4.0 is to have _some other_ (as in: Not the FPGA image you wish to test) FPGA image loaded. When the host application start it sees that the correct image is not loaded into FPGA, and starts loading your FPGA image - this works! 

Try it, and let me know if it fixes your issue.

Regards.
/Jesper

AWSaalluri, 2018-07-02:
Hi nikever, lyager,

Please accept our apologies for the inconvenience.  We had an issue where the default AFI was not accessible for us-west-2 & eu-west-1. We fixed it last week.  Could you please try again and let us know if the issue persists?

Thank you,
AWSaalluri