aws/aws-fpga #418: Create_Runtime_AMI.md is broken

vitesse-ftian, 2018-08-18:
Any issues related to HDK or SDK should be filed on FPGA development Forum.

The [AWS FPGA Development User Forum](https://forums.aws.amazon.com/forum.jspa?forumID=243) is the first place to go to post questions, suggestions and important announcements. 

Section 2,  Copy required Xilinx SDAccel Runtime Libraries to the Instance and Reboot your Runtime Instance.

The instruction is broken, first, it should be $XILINX_SDX, not $XIILNX_SDX.   Second, those binary files to be copied have be moved around in the newest developer AMI. 

Maybe just ask people to copy the rte.dyn directory?

vitesse-ftian, 2018-08-18:
On ubuntu, I have to do the following in addition to the instruction,

1. XILINX_SDX directory must be named as 2017.4.XXX, because the sdaccel_setup.sh uses first 6 char as release version,
2. Must copy lib/lnx64.0/Ubuntu directory as well.

After these steps seems the ubuntu f1 instance is working.  I can run helloworld.

AWScsaralay, 2018-08-20:
Hello,

Thank you for bringing this to our notice. The paths have changed in the 2017.4. We will update our documentation.

Thanks,
AWSChakra

AWSaalluri, 2019-01-25:
This has been fixed in Developer Kit release 1.4.6. Instruction should work for both 2017.4 & 2018.3.

https://github.com/aws/aws-fpga/blob/master/SDAccel/docs/Create_Runtime_AMI.md

Thanks,
AWSaalluri

closed 2019-01-25 by AWSaalluri