aws/aws-fpga #419: source sdaccel.sh fails on latest master

neelngc, 2018-08-21:
/home/centos/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_bo.c:55:9: error: too many arguments to function ‘drm_mm_insert_node_generic’
         return drm_mm_insert_node_generic(mm, node, size, PAGE_SIZE, 0, 0, 0);

AWSaalluri, 2018-08-22:
Hi 
Thank you for reporting this. we are working on a fix for this issue.

Please watch the AWS FPGA Developer's forum for updates:

https://forums.aws.amazon.com/forum.jspa?forumID=243&start=0

kindly report any future issues or post question about this repo on the forum.

Thanks,
AWSaalluri

AWSaalluri, 2018-08-23:
While we are working to implement a patch for this issue please refer to this forum announcement to avoid hitting this issue:

https://forums.aws.amazon.com/ann.jspa?annID=6060

Thank you,
AWSaalluri

closed 2018-08-23 by AWSaalluri