aws/aws-fpga #421: Vector Addition in OpenCL: need help

adki, 2018-08-23:
"Vector addition" (my own) causes segmentation fault on AWS.<br>

I prepared "vector addition" example, which was tested on local machine with KCU1500 board.
I moved the design to ASW F1 @ N. Virginia and then did as follows.

1. Set AWS FPGA environment
    ```$ source aws-fpga/sdaccel_setup.sh```
2. Prepared host program and xclbin for VCU1525
   ... as you know it took pretty long time ...
   ... it generates xclbin file someting like ```krnl_vadd.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin```
3. got root
    ```root$ sudo su```
4. Set run-time environment
   ```root$ source /opt/Xilinx/SDx/2017.4.rte.dyn/setup.sh```
5. Run the host program
    ```root$ ./host```
6. It ends up with following message
```
[0]user:0xf010:0x1d51:[xocl:2017.4.5:128]
xclProbe found 1 FPGA slots with xocl driver running
INFO: Importing xclbin/krnl_vadd.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
INFO: Loaded file
Segmentation fault (core dumped)
```
Do I need ```awsxclbin``` instead of ```xclbin```?

wilderfield, 2018-08-23:
When running on AWS you always need to refer to a registered AFI. The awsxclbin is essentially a pointer to an AFI. In ml-suite we drop the awsxclbin extension in favor of xclbin to keep things appearing uniform. Take a look at aws-fpga github repo for more details. Also please use forum for debug help.

kristopk, 2018-08-23:
Hi,
Please post questions to the AWS forum:
https://forums.aws.amazon.com/forum.jspa?forumID=243
Thanks!

closed 2018-08-23 by kristopk