aws/aws-fpga #424: create_sdaccel_afi.sh cannot connect to the S3 endpoint

adki, 2018-08-26:
I tried to make awsxclbin from xclbin using create_sdaccel_afi.sh, 
```
[centos@AWS] $SDACCEL_DIR/tools/create_sdaccel_afi.sh -xclbin=krnl_vadd.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin -o=krnl_vadd.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.awsxclbin -s3_bucket=sdaccel-fpga -s3_dcp_key=sdaccel-dcp -s3_logs_ke=sdaccel-log
```
but error occurs.
```
...
INFO: Split DCP from xclbin: 18_08_26-072403-primary.bit
INFO: Split Metadata from xclbin: 18_08_26-072403-primary.bit
INFO: Generated manifest file '18_08_26-072403_manifest.txt'
upload failed: ./18_08_26-072403_Developer_SDAccel_Kernel.tar to s3://sdaccel-fpga/sdaccel-dcp/18_08_26-072403_Developer_SDAccel_Kernel.tar Could not connect to the endpoint URL: "https://sdaccel-fpga.s3.us-east-1c.amazonaws.com/sdaccel-dcp/18_08_26-072403_Developer_SDAccel_Kernel.tar?uploads"
```

I have S3 bucket and DCP and LOG sub-directory on S3.
Thanks.

deeppat, 2018-08-28:
@adki , I believe you have set your default region to us-east-1c which is why the cli can't reach the endpoint. Amazon API endpoints are not AZ specific, so you should set it to the region. For eg, you should set it to us-east-1 and try again.

deeppat, 2018-09-04:
Closing this issue as I believe my comment should solve the problem. Please post on the forum for any follow ups!

-Deep

closed 2018-09-04 by deeppat

yunqu, 2020-03-06:
I have seen similar issue. Did @adki solve this problem by the comment? I used the us-east-1 as region but still no luck.

deeppat, 2020-03-06:
@yunqu, could you share the specific error message you saw?

yunqu, 2020-03-06:
I used the command:
```
./SDAccel/tools/create_sdaccel_afi.sh -xclbin=/home/centos/data/build/resizer.xclbin -o=resizer \
-s3_bucket=arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld \
-s3_dcp_key=arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld/dcps/ \
-s3_logs_key=arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld/logs/
```
Then it errors out with:
```
upload failed: ./20_03_06-191234_Developer_SDAccel_Kernel.tar to s3://arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld/arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld/dcps//20_03_06-191234_Developer_SDAccel_Kernel.tar Could not connect to the endpoint URL: "https://pynq-helloworld/arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld-186486331965.s3-accesspoint.us-east-1.amazonaws.com/pynq-helloworld%2Farn%3Aaws%3As3%3Aus-east-1%3A186486331965%3Aaccesspoint%2Fpynq-helloworld/dcps//20_03_06-191234_Developer_SDAccel_Kernel.tar?uploads"
```

yunqu, 2020-03-06:
@deeppat Ok I got it. The correct command should be:
```
./SDAccel/tools/create_sdaccel_afi.sh -xclbin=/home/centos/data/build/resizer.xclbin -o=resizer \
-s3_bucket=arn:aws:s3:us-east-1:186486331965:accesspoint/pynq-helloworld \
-s3_dcp_key=dcps/ \
-s3_logs_key=logs/
```
The dcp and logs section should not repeat the access point ARN.

deeppat, 2020-03-06:
@yunqu That's correct. internally, the script just calls aws s3 cp and only expects the S3Uri for the s3 bucket argument. I'm glad it's all set for you. I'll see if we can update the script to better specify the requirements too.