aws/aws-fpga #425: Asking for more specific example for generating AFI

adki, 2018-08-27:
Are anyone who can give more specific example for generating AFI.
I tried to generate AFI from xclbin. To do so, first I prepared S3 bucket, but it does occurs error; Setup CLI and Create S3 Bucket.

What I'm looking for is a real example for this.
```
   $ aws s3 mb s3://<bucket-name> --region us-east-1  # Create an S3 bucket (choose a unique bucket name)
    $ aws s3 mb s3://<bucket-name>/<dcp-folder-name>   # Create folder for your tarball files
    $ touch FILES_GO_HERE.txt                          # Create a temp file
    $ aws s3 cp FILES_GO_HERE.txt s3://<bucket-name>/<dcp-folder-name>/  # Which creates the folder on S3
```
What I did,
```
[centos@AWS] aws s3 mb s3://fpga-aws --region us-east-1
make_bucket: fpga-aws
[centos@AWS] aws s3 mb s3://fpga-aws/dcp
make_bucket failed: s3://fpga-aws/dcp Could not connect to the endpoint URL: "https://fpga-aws.s3.us-east-1c.amazonaws.com/"
[centos@AWS]
```

wilderfield, 2018-08-28:
What region is your ec2 instance running in? I suspect the bucket and the instance must be in the same region.

deeppat, 2018-08-28:
Hi @adki, it seems you've set your default region to us-east-1c.
You can check the default region in ~/.aws/config

Can you modify it to us-east-1 and try again.

deeppat, 2018-09-04:
Closing this issue. Please post on the forum for further follow-up.

closed 2018-09-04 by deeppat