aws/aws-fpga #428: source sdaccel_setup.sh fails

gsneha26, 2018-09-19:
CC [M]  /aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.o
In file included from /home/gsneha/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.c:19:0:
/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_drv.h:27:25: fatal error: drm/drm_gem.h: No such file or directory
 #include <drm/drm_gem.h>
                         ^
compilation terminated.
make[3]: *** [/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.o] Error 1
make[2]: *** [_module_/aws-fpga/sdk/linux_kernel_drivers/xocl] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.2.0-29-generic'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/aws-fpga/sdk/linux_kernel_drivers/xocl'
make: *** [all] Error 2
ERROR: Build of SDAccel runtime FAILED

kristopk, 2018-09-19:
Please provide your linux kernel version.
 %uname -r

gsneha26, 2018-09-19:
3.2.0-29-generic

rradjabi, 2018-09-19:
Can you please start from scratch by cloning aws-fpga and source sdaccel_setup.sh and then share the entire stdout?

gsneha26, 2018-09-20:
make -C /aws-fpga/sdk/linux_kernel_drivers/xocl
make[1]: Entering directory `/aws-fpga/sdk/linux_kernel_drivers/xocl'
echo "include: -I/aws-fpga/SDAccel/userspace/include -I/aws-fpga/sdk/linux_kernel_drivers/xocl -I/aws-fpga/sdk/linux_kernel_drivers/xocl/../xdma/"
include: -I/aws-fpga/SDAccel/userspace/include -I/aws-fpga/sdk/linux_kernel_drivers/xocl -I/aws-fpga/sdk/linux_kernel_drivers/xocl/../xdma/
echo "sdaccel_dir: /aws-fpga/SDAccel"
sdaccel_dir: /aws-fpga/SDAccel
echo "ROOT: "
ROOT:
echo "XOCL_DIR: /aws-fpga/sdk/linux_kernel_drivers/xocl"
XOCL_DIR: /aws-fpga/sdk/linux_kernel_drivers/xocl
make -C /lib/modules/3.2.0-29-generic/build M=/aws-fpga/sdk/linux_kernel_drivers/xocl modules
make[2]: Entering directory `/usr/src/linux-headers-3.2.0-29-generic'
  CC [M]  /aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.o
In file included from /aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.c:19:0:
/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_drv.h:27:25: fatal error: drm/drm_gem.h: No such file or directory
 #include <drm/drm_gem.h>
                         ^
compilation terminated.
make[3]: *** [/aws-fpga/sdk/linux_kernel_drivers/xocl/xocl_sysfs.o] Error 1
make[2]: *** [_module_/aws-fpga/sdk/linux_kernel_drivers/xocl] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.2.0-29-generic'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/aws-fpga/sdk/linux_kernel_drivers/xocl'
make: *** [all] Error 2
ERROR: Build of SDAccel runtime FAILED

aye20, 2018-09-21:
Can you please confirm what OS you are running on, it appears you are using an older kernel. 
Please refer to 
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_4/ug1238-sdx-rnil.pdf#page=25
for supported OSes for 2017.4.

kristopk, 2018-09-23:
Please re-open if you are seeing issues on the supported Ubuntu release (Ubuntu Linux 16.04.2).

closed 2018-09-23 by kristopk