aws/aws-fpga #446: aes_decrypt example

canadyre, 2018-12-10:
Hi,
I have been trying to work through the aes_decrypt example on an F1 instance but am having trouble getting it to run. I have been able to emulate it on software, hardware, also build it for hardware and even successfully go through the afi process. However, as soon as I try to run it, I get this output:

[centos@ip-172-31-63-235 aes_decrypt]$ sudo sh
sh-4.2# source /opt/xilinx/xrt/setup.sh 
XILINX_XRT      : /opt/xilinx/xrt
PATH            : /opt/xilinx/xrt/bin:/sbin:/bin:/usr/sbin:/usr/bin
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:
sh-4.2# ./aes -b data/test.txt -g data/goldFile.txt 
INFO: [main.cpp:39] TIME: [Mon Dec 10 03:36:08 2018] Xilinx AES-Decrypt Application started
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
device[0].user_instance : 0
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
device[0].user_instance : 0
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
../..//libs/xcl2/xcl2.cpp:40 Error calling err = cl::Platform::get(&platforms), error code is: -1001

Any help would be greatly appreciated!

AWScsaralay, 2018-12-17:
Thank you for bringing this to our attention. Xilinx has confirmed the issue and we'll have updates shortly.

rradjabi, 2018-12-20:
@canadyre 

The message "WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0" usually indicates the xocl driver isn't loaded properly. Please load with 'modprobe xocl'. At that point, 'awssak list' should indicate a device is present, else it will display that same message again. Please share your results.

deeppat, 2019-04-17:
@canadyre I'm closing this issue as I'm assuming this is resolved. Please feel free to reach out to us on our [forums](https://forums.aws.amazon.com/forum.jspa?forumID=243&start=0) if this is still failing for you.

closed 2019-04-17 by deeppat