aws/aws-fpga #457: Python bindings error in default AMI (undefined symbol: fpga_pci_readdir_mutex)

giech, 2019-04-15:
Thank you for adding a Python bindings API! I was excited to try it, but ran into an issue in `fpga_pci.py` when testing with AMI 1.6.0. Specifically, this is the traceback:

```
 import fpga_pci
  File "/usr/lib64/python2.7/site-packages/fpga_pci.py", line 196, in <module>
    fpga_pci_readdir_mutex = (union_c__UA_pthread_mutex_t).in_dll(_libraries['/usr/local/lib64/libfpga_mgmt.so'], 'fpga_pci_readdir_mutex')
ValueError: /usr/local/lib64/libfpga_mgmt.so: undefined symbol: fpga_pci_readdir_mutex
```

The issue stems from a conditional compilation in [fpga_pci.h](https://github.com/aws/aws-fpga/blob/0f678055fa9f1378610389e46750f8607a0d4c23/sdk/userspace/include/fpga_pci.h#L246), which affects the existence of `fpga_pci_readdir_mutex` in [fpga_pci.py](https://github.com/aws/aws-fpga/blob/6c707ab4a26c2766b916dad9d40727266fa0e4ef/sdk/userspace/python_bindings/fpga_pci.py#L196). 

As a side-note, the AMI ships with a `libc` version of 2.17, so `readdir_r ` is available:
```
ldd --version
ldd (GNU libc) 2.17
```

Commenting line 196 in `fpga_pci.py` allows me to run code as needed, so asolution would be to conditionally include or remove `fpga_pci_readdir_mutex` in Python (and the `__all__` variable in the same file).

deeppat, 2019-04-17:
Hello, thanks for reporting this issue! 
We already have a fix that we are testing internally and will release it with the next major release(1.4.9)

giech, 2019-04-21:
Thank you for confirming! Since this is somewhat too minor for a separate bug, can you also take a look at https://github.com/aws/aws-fpga/blob/master/sdk/apps/byte_swapper/fpga_funcs.py#L64? I think `0` should be replaced by `self.slot`.

deeppat, 2019-05-16:
Hi @giech Thanks for pointing that out as well. 

On a side note, our team is still working on a fix and while we could not get it released in time for 1.4.9, I'm targeting it for the next rev. I'll update you here once we have the fixes out! 

Thanks again,

Deep

deeppat, 2021-01-12:
This fix was released. 
Closing.

closed 2021-01-12 by deeppat