aws/aws-fpga #461: Runtime packages

xlz-jgoutin, 2019-06-04:
Please, consider creating .deb/.rpm packages for runtime libraries and utilities (Ex `aws-fpga-mgmt`) for most common OS (Ubuntu/CentOS).

Cloning the git and sourcing the script is OK for development environments, but much less for a runtime environment. Using packages may help to:
- Reduce downloading from full source (~275MB) to only required compiled files (~0.3MB).
- Remove the need to install build requirements (GCC, Git, ...).
- Reduce the whole install duration.
- Handle install and updates using the package manager.
- Allow signature verification.

The package may contain:

Management utilities (Maybe not all if some are not relevant in runtime environment):
- `/usr/bin/fpga-clear-local-image`
- `/usr/bin/fpga-describe-local-image`
- `/usr/bin/fpga-describe-local-image-slots`
- `/usr/bin/fpga-get-virtual-dip-switch`
- `/usr/bin/fpga-get-virtual-led`
- `/usr/bin/fpga-load-local-image`
- `/usr/bin/fpga-local-cmd`
- `/usr/bin/fpga-set-virtual-dip-switch`
- `/usr/bin/fpga-start-virtual-jtag`

Libraries:
- `/usr/lib64/libfpga_mgmt.so`
- `/usr/lib64/libfpga_mgmt.so.1`
- `/usr/lib64/libfpga_mgmt.so.1.0.0`

Udev rule:
- The script to generate udev rule, `sdk/userspace/add_udev_rules.sh` (By example as `/usr/bin/fpga-add-udev-rule`)
- `/opt/aws/bin/change-fpga-perm.sh` (Maybe moved to `/usr/bin/fpga-change-perm`)

Eventually, a development package (Example: `aws-fpga-mgmt-dev` deb package  / `aws-fpga-mgmt-devel` rpm package) may also be created to install include files (from `/sdk/userspace/include`) to allow build software without the whole SDK.

deeppat, 2019-06-06:
Hi @xlz-jgoutin, this is excellent feedback for us to take back to our development team. 

This adds a lot of value and ease of use for a runtime environment. I will discuss it with the development team and discuss where we can place it within our development efforts. I'll update here once I have something for you. 

-Deep

kyyalama2, 2022-04-06:
Thank you so much for the feedback. We will post the future improvements and updates once available. Thanks

closed 2022-04-06 by kyyalama2