aws/aws-fpga #469: The "cl_dram_dma" pre-gen AFI can not be load while "cl_hello_world" AFI can be load.

LeonMac, 2019-10-11:
Hello, I am doing some quick test by using the pre-generated AFI, so I don't need rebuild it.

I can find "cl_hello_word" AFI: agfi-0fcf87119b8e97bf3 (https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_hello_world/README.md#metadata).
This AFI can be loaded correctly.

I can find "cl_dram_dma" AFI: agfi-02f141212beac0cfb (https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_dram_dma).
However this AFI can not be loaded:

Can someone have a check on that?
`ubuntu@ip-:/aws-fpga/hdk/cl/developer_designs/cl_zcash/software/runtime$ sudo fpga-load-local-image -S 0 -I agfi-0fcf87119b8e97bf3
AFI          0       agfi-0fcf87119b8e97bf3  loaded            0        ok               0       0x04261818
AFIDEVICE    0       0x1d0f      0xf000      0000:00:1d.0
ubuntu@ip-: /aws-fpga/hdk/cl/developer_designs/cl_zcash/software/runtime$ sudo fpga-clear-local-image -S 0 
AFI          0       none                    cleared           1        ok               0       0x04261818
AFIDEVICE    0       0x1d0f      0x1042      0000:00:1d.0
ubuntu@ip-: /aws-fpga/hdk/cl/developer_designs/cl_zcash/software/runtime$ sudo fpga-load-local-image -S 0 -I agfi-02f141212beac0cfb
Error: (5) invalid-afi-id
The agfi id passed is invalid or you do not have permission to load
the AFI.
ubuntu@ip-: /aws-fpga/hdk/cl/developer_designs/cl_zcash/software/runtime$ `

deeppat, 2019-10-11:
Hi @LeonMac, the AFI should be available in all GA regions. Can you confirm the region you see this error in?

LeonMac, 2019-10-12:
Yes I believe so as this is an example AFI. But as you can see there are some error report for this AFI.
I am in China region: cn-north-1

deeppat, 2019-10-14:
@LeonMac The error is because we haven't replicated the cl_dram_dma AFI's to all preview regions. Preview regions might not have the same availability as GA regions. I'll see if we can get the AFI's available in China and get back to you.

LeonMac, 2019-10-15:
@deeppat Thanks for help. I assume few copy command can make this available in China region--but this can really help people saving a lot of resource if they don't want to rebuild the code (I had done the cl_helloworld build, take more than 5 hrs--really make no sense to rebuild that example project if one already did practice..

deeppat, 2021-01-12:
Closing as AFI's are not guaranteed to be available in preview regions.

closed 2021-01-12 by deeppat