aws/aws-fpga #472: Missing <linux/version.h> in mgmt-core.h cause failure to compile DKMS on kernel 3.10.0-1062.1.2

mdfck, 2019-10-18:
After updating the kernel to 3.10.0-1062.1.2 and installing the respective kernel-headers and kernel-devel packages, dkms autoinstall will fail to compile the awsmgmt and xocl modules.

### Errors

`# dkms autoinstall`
``
`Kernel preparation unnecessary for this kernel.  Skipping...`
``
...
`Error!  Build of awsmgmt.ko failed for: 3.10.0-1062.1.2.el7.x86_64 (x86_64)`
`Consult the make.log in the build directory`
...
`# cat /var/lib/dkms/xrt-aws/2.2.0/build//make.log`
`In file included from /var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt/mgmt-core.c:25:0:`
`/var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt/mgmt-core.h:28:5: warning: "LINUX_VERSION_CODE" is not defined [-Wundef]`
`...`
`/var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt/mgmt-core.h:28:27: warning: "KERNEL_VERSION" is not defined [-Wundef]`
`...`
`/var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt/mgmt-core.h:28:41: error: missing binary operator before token "("`
`...`
`make[2]: *** [/var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt/mgmt-core.o] Error 1`
`make[1]: *** [_module_/var/lib/dkms/xrt-aws/2.2.0/build/driver/aws/kernel/mgmt] Error 2`
`make[1]: Leaving directory '/usr/src/kernels/3.10.0-1062.1.2.el7.x86_64'`
`make: *** [all] Error 2`

### Workaround (fix?)
I have managed to compile the module by adding the include line to the mgmt-core.h file:

`# pwd`
`/usr/src/xrt-aws-2.2.0/driver/aws/kernel/mgmt`
`# diff -c mgmt-core.h mgmt-core.h.bkp`
`*** mgmt-core.h	2019-10-18 07:31:14.254081701 +0000`
`--- mgmt-core.h.bkp	2019-09-06 19:46:17.000000000 +0000`
`***************`
`*** 15,21 ****`
`  #define _AWS_MGT_PF_H_`
``
`  #include <linux/cdev.h>`
`- #include <linux/version.h>`
`  #include <asm/io.h>`
``  
`  #define DRV_NAME "awsmgmt"`
`--- 15,20 ----`

deeppat, 2019-10-24:
Hello @mdfck Could you tell us what AMI version this is an issue in? We will reach out to Xilinx and ask them for an official fix

mdfck, 2019-10-24:
Hi @deeppat,

I have used the FPGA Developer AMI v1.7.0 from Marketplace. XRT 2.2.0. That AMI comes with an older kernel version. Running "yum -y groupinstall "GNOME Desktop"" will resolve some dependencies and install a VDO package that requires kernel > 3.10.0-1025, and trigger the kernel update.

Immediately after the update, I rebooted the instance and the xocl driver was no longer available. 

During the yum command there are error messages stating that the kernel-headers and kernel-devel packages were not available in the correct version (because the Gnome installation did not trigger those), but once I updated those packages and ran "dkms autoinstall", the compilation failed with the messages I provided initially.

Double checking, I noticed that the output of diff is garbled. I just had to add the #include for linux/version.h and run "dkms autoinstall".

The file I changed was from the aws-fpga package, hence I reported the issue here.

deeppat, 2021-01-12:
This was fixed with new XRT updates. Closing.

closed 2021-01-12 by deeppat