aws/aws-fpga #480: source sdaccel_setup.sh failing

cathieO, 2020-02-29:
Following your instructions we are not able to source sdaccel_setup.sh.
It appears that the problem is that sdx is not installed on the aws instance.
We are running a f1.2xlarge aws instance.

The error message is 
ERROR: XILINX_SDX variable not set and sdx not in path
ERROR: Please set XILINX_SDX variable to point to your location of your Xilinx installation or add location of sdx executable to your PATH variable.

However, we are unable to locate sdx on the system to do either of those things.

Thanks in advance

deeppat, 2020-02-29:
Hello,

The answer to this question will  really depend on the AMI of the instance you are running the setup script on.

XILINX_SDX is already set on the FPGA Developer AMI versions till v1.7.0. If you are using FPGA Developer AMI v1.8.0, it should be using the Vitis flow instead of the SDAccel flow and you'd have to source the vitis_setup.sh script instead. 

If you are using a Custom image or on-premise instances, you might have to set the XILINX_SDX variable to the install path.

Hope this helps and feel free to respond here if yours is a special case and we'll help you through this and improve our documentation to include this case too.

Thanks,
Deep

cathieO, 2020-03-02:
Thank you for your response.
The documentation at
https://github.com/aws/aws-fpga
still points users to the SDAccel HelloWorld example to get started.

Maybe, https://github.com/aws/aws-fpga/tree/master/Vitis would be an alternative that could prevent this confusion?

deeppat, 2020-03-02:
@cathieO Thank you! That's really good feedback. 

I'm going to make some minor updates today based on that and we will be making more updates on the documentation to move the emphasis over from SDAccel to Vitis overall. I'll close this issue for now, but feel free to ask us about any other issues/feedback you may have!

-Deep

closed 2020-03-02 by deeppat