aws/aws-fpga #485: Error starting instances

QueSSence, 2020-04-17:
[image]

deeppat, 2020-04-17:
@QueSSence: Depending on customer usage in a specific AZ or region, one can see such errors. Please look at the [EC2 troubleshooting guide](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-launch.html#troubleshooting-launch-capacity) as it explains next steps.

QueSSence, 2020-04-19:
Hi Deeppat, Now I am getting a different error, we generating a bit file within specified limits
[image]

rakesh-ceremorphic, 2020-04-19:
I am trying to launch f1.2xLarge instance.But getting above errors.I checked the documentation page.This configuration is supported..Please let me know what could be the issue.

deeppat, 2020-04-21:
Hi @rakesh-ceremorphic @QueSSence , I think you'd be better served if you start a support ticket with AWS. We are here to help you out with the Developer Kit and I feel like AWS Support would be much better equipped to help you out with EC2 issues like this. Please let us know if you have further questions that we can help with!

deeppat, 2020-09-07:
Closing as I am assume this problem is resolved. Please re-open if this is still an issue!

closed 2020-09-07 by deeppat