aws/aws-fpga #491: Outdated XDMA drivers

jrtc27, 2020-06-26:
The official Xilinx XDMA drivers have seen recent updates, both for compatibility with "newer" kernel versions (even those old enough to be in Ubuntu 18.04) and highly important bug fixes. In particular, https://github.com/Xilinx/dma_ip_drivers/commit/d1f334b1372dbd7f2fa6f28a31b496779119f8d7 meant that, if the CL was too slow at serving the DMA engine's requests (but still orders of magnitude below the timeouts), the XDMA driver would sleep for 10s and then report a timeout, despite the request actually completing. Our CL was slow enough that this happened under contention. Please update to this latest version; adding the AWS F1 PCI IDs to the table is all that's required for them to work.

deeppat, 2020-06-26:
Hello, thanks for bringing this up. I'll sync up with the team and Xilinx on updating the DMA drivers.

jrtc27, 2020-06-30:
Hi, thanks for the quick response. Whilst you're doing that, can I please draw https://github.com/Xilinx/dma_ip_drivers/pull/49 to the attention of you and Xilinx? Especially my comment (https://github.com/Xilinx/dma_ip_drivers/pull/49#issuecomment-651435242) which provides what should be a more amenable patch (minus the BEGIN/END comments to denote local changes in our vendored copy) than the existing one that is currently sitting there in review. Having fixed the timeout issue I then ran into this today, and reading the current sources in aws-fpga it's a long-standing bug that's still present as far back as those.

deeppat, 2020-07-02:
Hi @jrtc27,

Xilinx asked me to check if you are using the latest commit: The issue should already be fixed by commit [5faf23e](https://github.com/Xilinx/dma_ip_drivers/commit/5faf23ec45794c82d4e1bfb283d25934a790cca1).

Would you be able to confirm if building with that sha works for you?

jrtc27, 2020-07-02:
Yes, it did, thank you (I replied to Karen at https://github.com/Xilinx/dma_ip_drivers/pull/49#issuecomment-651985365).

deeppat, 2020-09-07:
We'll include the updates from Xilinx in our next releases. Closing this issue

closed 2020-09-07 by deeppat