aws/aws-fpga #492: host: './vadd.awsxclbin' is not a legal IDN name (Output would be too large or too small), use +noidnin

jelling, 2020-06-26:
I followed the Hello World Readme but this error when I try to run the FPGA binary:

```
[centos@ip-172-31-81-91 aws-fpga]$ pwd
/home/centos/aws-fpga
[centos@ip-172-31-81-91 aws-fpga]$ host ./vadd.awsxclbin
host: /opt/Xilinx/Vitis/2019.2/lib/lnx64.o/libxml2.so.2: no version information available (required by /lib64/libdns.so.1102)
host: /opt/Xilinx/Vitis/2019.2/lib/lnx64.o/libxml2.so.2: no version information available (required by /lib64/libisc.so.169)
host: './vadd.awsxclbin' is not a legal IDN name (Output would be too large or too small), use +noidnin
[centos@ip-172-31-81-91 aws-fpga]$ 
```

Also, the Readme specifies to run the binary using `./host ./vadd.awsxclbin` but pretty sure it shouldn't be specifying the local directory for the binary source.

deeppat, 2020-06-26:
Hi @jelling, 

The `host` executable to use in this case is the one compiled during the build stage: 
```
    $ cd $VITIS_DIR/examples/xilinx/hello_world           
    $ make clean                                                             
    $ make TARGET=hw DEVICE=$AWS_PLATFORM all --> will produce a host executable
```
The documentation assumes that the user built the CL/AFI/awsxclbin and host executable in a C5/Z1d build instance and are copying those to an F1 instance using the FPGA Developer AMI. I apologize for the confusion in this case and we will update it to be more concise.

After you've copied them to your F1 instance , you can run the `./host ./vadd.awsxclbin` from the location where you have copied both artifacts to.

Please let me know if you still have issues with running the application on F1 and I'll be happy to help!

-Deep

jelling, 2020-07-08:
Thank you! My mistake.

deeppat, 2020-07-10:
Ok, I'll close this issue for now. Please feel free to ask us anything F1 here!

closed 2020-07-10 by deeppat