aws/aws-fpga #501: create_sdaccel_afi.sh cannot connect to the S3 endpoint.

liyewen521, 2020-09-11:
Hello, I tried to make awsxclbin from xclbin using create_sdaccel_afi.sh. And I use the below command when I have read past issues:

`$SDACCEL_DIR/tools/create_sdaccel_afi.sh -xclbin=BSW.hw.xclbin -o=BSW.hw \
        -s3_bucket=arn:aws:s3:us-east-2:735077837443:accesspoint/wga -s3_dcp_key=dcp -s3_logs_key=logs`

but I got those error messages:

`INFO: Generated manifest file '20_09_11-112011_manifest.txt'
upload failed: ./20_09_11-112011_Developer_SDAccel_Kernel.tar to s3://arn:aws:s3:us-east-2:735077837443:accesspoint/wga/dcp/20_09_11-112011_Developer_SDAccel_Kernel.tar Parameter validation failed:
Invalid bucket name "arn:aws:s3:us-east-2:735077837443:accesspoint": Bucket name must match the regex "^[a-zA-Z0-9.\-_]{1,255}$"`

Can you give me some help? Thank you very much!

deeppat, 2020-09-11:
I think the only thing you need to change is: 

`$SDACCEL_DIR/tools/create_sdaccel_afi.sh -xclbin=BSW.hw.xclbin -o=BSW.hw -s3_bucket=**s3://**arn:aws:s3:us-east-2:735077837443:accesspoint/wga -s3_dcp_key=dcp -s3_logs_key=logs`

It should work after that.

liyewen521, 2020-09-12:
I used this command and got the below message:

`upload failed: ./20_09_12-020143_Developer_SDAccel_Kernel.tar to s3://s3://arn:aws:s3:us-east-2:735077837443:accesspoint/wga/dcp/20_09_12-020143_Developer_SDAccel_Kernel.tar Parameter validation failed:
Invalid bucket name "s3:": Bucket name must match the regex "^[a-zA-Z0-9.\-_]{1,255}$"`

According to this message, "s3://" may need to be removed, can you give me some help?

deeppat, 2020-09-12:
Ok this seems to be due to an older AWS CLI that does not recognize access points.

I'm unable to get a new cli from yum, however the following steps should get you a new aws cli working with your original command:

```
# Remove old cli
sudo yum remove awscli
curl "https://s3.amazonaws.com/aws-cli/awscli-bundle.zip" -o "awscli-bundle.zip"
unzip awscli-bundle.zip
sudo ./awscli-bundle/install -i /usr/local/aws -b /usr/local/bin/aws

# Verify version
aws --version
aws-cli/1.18.137 Python/2.7.5 Linux/3.10.0-1062.4.1.el7.x86_64 botocore/1.17.60

# Test upload..you can replace with your own create_sdaccel_afi command
aws s3 cp vector_addition.xclbin s3://arn:aws:s3:us-east-1:719506761471:accesspoint/test-ap/dcps/
upload: ./vector_addition.xclbin to s3://arn:aws:s3:us-east-1:719506761471:accesspoint/test-ap/dcps/vector_addition.xclbin

$SDACCEL_DIR/tools/create_sdaccel_afi.sh -xclbin=BSW.hw.xclbin -o=BSW.hw -s3_bucket=arn:aws:s3:us-east-2:735077837443:accesspoint/wga -s3_dcp_key=dcp -s3_logs_key=logs 
```

Please let me know if this doesn't work for you.

liyewen521, 2020-09-12:
Thanks for your advice, now I can upload data to s3 when running `aws s3 cp ...`
But when I run 

`$SDACCEL_DIR/tools/create_sdaccel_afi.sh -xclbin=BSW.hw.xclbin -o=BSW.hw -s3_bucket=arn:aws:s3:us-east-2:735077837443:accesspoint/wga -s3_dcp_key=dcp -s3_logs_key=logs`

I got the following error message since it loads Kernel.tar successfully:

`upload: ./20_09_12-041631_Developer_SDAccel_Kernel.tar to s3://arn:aws:s3:us-east-2:735077837443:accesspoint/wga/dcp/20_09_12-041631_Developer_SDAccel_Kernel.tar
An error occurred (InternalError) when calling the CreateFpgaImage operation (reached max retries: 4): An internal error has occurred`

How to fix this problem?

deeppat, 2020-09-13:
Hi @liyewen521, what region are you calling it in? Is it in us-east-2?

liyewen521, 2020-09-14:
My f1.2xlarge instance is at us-east-1 and the s3 seems at s3:us-east-2, should they in the same region?

deeppat, 2020-09-14:
@liyewen521 Thanks for trying the us-east-1 bucket out! I will check with our team to see why that gave an Internal Error earlier.

This error could be due to the AFI not being able to load. Could you verify the status of your AFI that you created earlier when you called `$SDACCEL_DIR/tools/create_sdaccel_afi.sh`?

Another way is to load the AFI directly using the SDK:

```
#Assuming your .awsxclbin file = BSW.hw.awsxclbin
strings BSW.hw.awsxclbin | grep agfi 

#Use the AFI output to try and load it:
#For eg:
sudo fpga-load-local-image -S0 -I <AGFI from strings output>
```

Can you verify what region the AFI was created in? For eg, what was the region set in your aws credentials file when you ran the `$SDACCEL_DIR/tools/create_sdaccel_afi.sh` script? If that was set to us-east-2, your AFI will need to be copied to us-eat-1 before using it on an F1 instance.

liyewen521, 2020-09-15:
I try to follow the steps you gave, and the messages are as following:
` strings BSW.hw.awsxclbin | grep agfi ` 
agfi-0e4fe8a1b347c6658
` sudo fpga-load-local-image -S0 -I agfi-0e4fe8a1b347c6658 `
Error: (5) invalid-afi-id
The agfi id passed is invalid or you do not have permission to load the AFI.

deeppat, 2020-09-15:
@liyewen521 

1. It takes ~ 30 mins or so to create an AFI after the call has been made. So an AFI might show as invalid if you call it right after you make the create-fpga-image call(call made from the create_sdaccel_afi script).
2. Once the AFI has been created, you can check the AFI ID in the afi_ids file and run an ec2 api call to verify that it is available: ```aws ec2 describe-fpga-images --region us-east-1 --fpga-image-id afi-xxxxx``` Note that AFI is the regional identifier for an FPGA image, and an AGFI is a global id.
3. AFI's by default are only accessible by the AWS account that has created them. If you used credentials vended by account A, can you make sure your F1 instance is launched by account A too. Or you can modify [AFI permissions](https://docs.aws.amazon.com/cli/latest/reference/ec2/modify-fpga-image-attribute.html)

I think you should be able to load the AFI now in your instance if the instance is launched by the same account. You do not need to call create_sdaccel_afi again for that design. Just call `sudo fpga-load-local-image -S0 -I agfi-0e4fe8a1b347c6658` or run your application .

liyewen521, 2020-09-17:
@deeppat Okay, the codes can run successfully now. Well, running code in old AMI is not a good choice, I will use the latest AMI next time. Thanks for your patient and useful help, I think this issue can be close now!

deeppat, 2020-09-21:
Great! Thanks @liyewen521!

closed 2020-09-21 by deeppat