aws/aws-fpga #506: How to install GUI after I have successfully run the hello world

tangyuelm, 2020-10-28:
Any issues related to HDK or SDK should be filed on FPGA development Forum.

The [AWS FPGA Development User Forum](https://forums.aws.amazon.com/forum.jspa?forumID=243) is the first place to go to post questions, suggestions and important announcements. 
Hello,
I am new to aws and want to use the f1 instance to complete Vivado block design with HLS iP and running it on sdk. I have followed the instructions on https://github.com/aws/aws-fpga/tree/master/hdk#gettingstarted and successfully ran Hello_world design.
I also want to use the GUI and try your DMA HLx example. I followed the instructions on https://github.com/aws/aws-fpga/blob/master/developer_resources/DCV_with_ParallelCluster.md for NICE DCV. When I configure AWS CloudFormation, there is not VPC IP options, but only DB options. I tried to follow but failed. Is there any explicit instructions for me to access GUI? I have already launched a f1 x2large instance and completed all configuration to run the Hello World example.

Thanks

AWSbatsunil, 2020-10-29:
Hi tangyuelm,

I think you can setup a dcv server on your current f1 instance.
You can follow the instructions mentioned in this link
[
https://github.com/aws/aws-fpga/blob/master/developer_resources/DCV.md
](
https://github.com/aws/aws-fpga/blob/master/developer_resources/DCV.md
)

Let us know if it solves the purpose.

Thanks,
Sunil.

tangyuelm, 2020-10-29:
Thanks for your reply. I just changed into RDP.

tangyuelm, 2020-10-29:
But now I have a new problem. I tried to run the IPI CDMA Test Example and followed the instructions from https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_ipi_cdma_test_hlx/README.md
When building the block diagram, it shows an error of ERROR: [Vivado 12-172] File or Directory '/home/centos/src/project_data/aws-fpga/hdk/common/shell_v04261818/design/ip/axi_register_slice/sim/axi_register_slice.v' does not exist
I also tried the cl_hello_world_ref_hlx and got a similar error. Could you please tell me how to solve it and why I have such a problem?

tangyuelm, 2020-10-29:
I also got such error: [runtcl-4] can't read "::env(FAAS_SCRIPT_DIR)": no such variable
when I run synthesis. Could any one help me to fix the problem?

deeppat, 2020-10-29:
Hi @tangyuelm, I started with the FPGA Developer AMI v1.9.0 and ran the following steps:
```
git clone https://github.com/aws/aws-fpga
cd aws-fpga
source hdk_setup.sh

...

mkdir -p ~/.Xilinx/Vivado
echo 'source $::env(HDK_SHELL_DIR)/hlx/hlx_setup.tcl' >> ~/.Xilinx/Vivado/Vivado_init.tcl 
cd hdk/cl/examples/cl_ipi_cdma_test_hlx

vivado & 

# Next steps are in the tcl console in Vivado gui:
aws::make_ipi -examples cl_ipi_cdma_test

```

This ended up creating the the block diagram correctly. 

I believe the problem might have been the missing setup script? 

Can you make sure you have run the following steps:

```
source hdk_setup.sh
mkdir -p ~/.Xilinx/Vivado
```

Then verify that the `~/.Xilinx/Vivado/Vivado_init.tcl` file has the following text: `source $::env(HDK_SHELL_DIR)/hlx/hlx_setup.tcl`

If not, run the below command:
```
echo 'source $::env(HDK_SHELL_DIR)/hlx/hlx_setup.tcl' >> ~/.Xilinx/Vivado/Vivado_init.tcl
```

After setting these env vars, start the vivado gui from the same terminal(to maintain environment variables). It should be able to then run the `make_ipi` command.

Let us know if you still can't make progress here.

-Deep

tangyuelm, 2020-10-29:
Hi, Deep,
Thanks for your reply. I indeed set up the HLx before. When I delete echo 'source $::env(HDK_SHELL_DIR)/hlx/hlx_setup.tcl' >> ~/.Xilinx/Vivado/Vivado_init.tcl, the building cannot start, while my error occurs during the middle of building the block.
I restart from the beginning of git clone https://github.com/aws/aws-fpga and now the whole design can be built. I thought it may have some problems when I extending the volume this afternoon. Hope the following flow can succeed. But anyway, thank you very much.

deeppat, 2020-10-30:
@tangyuelm Great. I'll close this issue for now, but please re-open if it doesn't work for you!

-Deep

closed 2020-10-30 by deeppat

tangyuelm, 2020-10-31:
Hello,

I found this problem happen every time when I open vivado the GUI. Even though I reopen and re-synthesis the project that I have implemented before, the error will come. Do I need to start from the git clone every time I need to run the IP integration flow?

AWSbatsunil, 2020-11-02:
Hi tangyuelm,

The git clone is not needed every time. If you open a new terminal window you need to source hdk_setup.sh for that terminal.

Let us know if it solves the problem. If you still see the issue can you please describe all the steps that you are following so that I can try to recreate the same steps that you are following.

Thanks,
Sunil.

tangyuelm, 2020-11-04:
Hi Sunil,

Thanks for your reply. It seems that updating the IP can work, but I am not quite sure now. I want to test my own program using the IP integration flow. It can synthesize and generate checkpoints, but I have not found ways to run it using the SDK application project.