aws/aws-fpga #508: XILINX_VITIS varaiblet not set

yanyr, 2020-10-31:
Hi,

after I source vitis_setup.sh a get these errors in the end:
ERROR: XILINX_VITIS variable not set and vitis not in the path
ERROR: Please set XILINX_VITIS variable to point to your location of your Xilinx installation or add location of vitis exectuable to your PATH variable

And I also cannot find Xilinx folder in the examples.

AWSbatsunil, 2020-11-02:
Hi yanyr,

Can you please let us know the AMI that you are running the command on?

[FPGA Developer AMI](https://aws.amazon.com/marketplace/pp/B06VVYBLZZ) should be used to do FPGA related developments.

Thanks,
Sunil.

yanyr, 2020-11-02:
Hi Sunil,

the latest 1.7.1

Yan

deeppat, 2020-11-02:
Hi @yanyr 1.7.1 is an updated version of the AMI with Xilinx toolset 2019.1 which does not include Vitis as Vitis was not launched till Xilinx 2019.2 tool release. To use Vitis, please use AMI v1.8.1 or 1.9.0 from the AWS Marketplace dropdown menu.

deeppat, 2020-11-10:
Closing this issue. Please re-open if you weren't able to use v1.9.0!

Thanks,
Deep

closed 2020-11-10 by deeppat

ashwintcs, 2023-03-28:
Hi
Just to confirm if we are following on-premise, do we still need to install Vitis on F1 instance?

kyyalama2, 2023-03-28:
If your design uses vitis flow, then you do need XRT envrionment and thsi should come by default if you are using one of the AWS provided AMIs. Thanks