aws/aws-fpga #509: Cannot load awsxclbin to F1 instance

Hecmay, 2020-11-07:
Hi, I was following the instructions here to create and load AFI. I was able to create an AFI image and copy it to F1 instance when it is done processing. However, when running the host program, I got the following error:

```shell
[centos@ip-172-31-53-85 kmeans]$ ./host
Found Platform
Platform Name: Xilinx
INFO: Reading kernel.awsxclbin
Loading: 'kernel.awsxclbin'
Trying to program device[0]: xilinx_aws-vu9p-f1_dynamic_5_0
XRT build version: 2.6.0
Build hash: 12115fd4054cb46a5ade62fafa74c523f59116e6
Build date: 2020-07-09 15:20:59
Git branch: 2020.1
PID: 2664
UID: 1000
[Fri Nov  6 18:41:07 2020]
HOST: ip-172-31-53-85.ec2.internal
EXE: /home/centos/alveo-test/kmeans/host
[XRT] ERROR: See dmesg log for details. err=-22
[XRT] ERROR: Failed to load xclbin.
Failed to program device[0] with xclbin file!
Failed to program any device found, exit!
```

The log from dmesg:
```shell
[  562.891743] xocl 0000:00:1d.0: icap.u.16777216 ffff88633f87a410 get_axlf_section_hdr: could not find section header 20
[  562.945226] xocl 0000:00:1d.0: icap.u.16777216 ffff88633f87a410 __icap_peer_xclbin_download: peer xclbin download err: -22
[  562.978334] xocl 0000:00:1d.0:  ffff88633edcf098 xocl_read_axlf_helper: Failed to download xclbin, err: -22
``` 

It seems that the xclbin cannot be downloaded successfully. Any idea how this can be fixed?

deeppat, 2020-11-09:
Hi @Hecmay, It seems like you copied the .awsxclbin file to an F1 instance. Did you wait till the AFI was created? AFI generation takes ~30 mins or so after the create_vitis_afi.sh script is called. To check whether the AFI is created, could you verify: https://github.com/aws/aws-fpga/tree/master/Vitis#tracking-the-status-of-your-registered-afi

In this case it seems like it failed to load the AFI on the instance. Another thing to check is whether the F1 instance was on the same AWS account as the AWS credentials used on the instance where the create_vitis_afi.sh was called. If it is a separate account, you need to provide AFI load permissions: https://github.com/aws/aws-fpga/blob/master/hdk/docs/fpga_image_attributes.md

After you have verified both, you should be able to run the host code. The host program looks at the .awsxclbin file, loads the AFI and then runs your accelerator. 

Hope this helps. Feel free to respond here if this didn't fix your issue.

-Deep

ThomasXilinx, 2020-11-09:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi Deep,

This issue reminds of an issue which I delt with in this thread:
https://forums.aws.amazon.com/thread.jspa?messageID=957617&#957617

After a lot of head scratching, it turned out to be that the user hadn’t configured the AWS CLI to output JSON.
This is why the .xclbin wasn’t getting loaded.
Perhaps you can ask this user about this.
Also, it would be nice if the create_vitis_afi.sh script checked for the AWS CLI output setting.

  Thomas

deeppat, 2021-01-12:
Hi @Hecmay I'm closing this as this thread was idle for a while. Please reopen if this is still an issue.

closed 2021-01-12 by deeppat