aws/aws-fpga #513: source sdaccel_setup.sh on FPGA DEV AMI

mcousine, 2021-02-20:
Done with SDK install.
INFO: sdk_setup.sh PASSED
ERROR: XILINX_VITIS variable is set, but you are calling sdaccel_setup.sh. This likely means that you are calling source sdaccel_setup.sh with Xilinx Vitis installed. Xilinx has replaced SDAccel with Vitis from 2019.2 release onwards. Please checkout the Vitis README and flow instead.

deeppat, 2021-02-22:
@mcousine Are you trying to build SDAccel based accelerators?

deeppat, 2021-06-23:
Closing due to inactivity. @mcousine, please re-open and we can help out further

closed 2021-06-23 by deeppat

falken42, 2024-07-09:
For anyone else who is just beginning to learn how to work with FPGAs (like me) and happens to run into this same error, follow the [Vitis README.md](https://github.com/aws/aws-fpga/blob/master/Vitis/README.md) and run `source vitis_setup.sh` instead.