aws/aws-fpga #517: Crash on dsaccel.sh , double declaration?

inemp, 2021-05-05:
Seems like it fpga_pci_get_address is declared twice? Is this a known error?

fpga_pci.c: At top level:
fpga_pci.c:455:1: error: conflicting types for ‘fpga_pci_get_address’
 fpga_pci_get_address(pci_bar_handle_t handle, uint64_t offset, size_t len,
 ^
In file included from fpga_pci_internal.h:20:0,
                 from fpga_pci.c:29:
../../include/fpga_pci.h:229:5: note: previous declaration of ‘fpga_pci_get_address’ was here
 int fpga_pci_get_address(pci_bar_handle_t handle, uint64_t offset,
     ^
make: *** [fpga_pci.o] Error 1
make failed
Error: mkall_fpga_mgmt_tools.sh returned 1
Error: AWS SDK install was unsuccessful, sdk_install.sh returned 0

AWScsaralay, 2021-05-05:
Hi, 

Did you follow instructions described in https://github.com/aws/aws-fpga/tree/master/sdk#quick-start ?

Could you also specify which AMI are you using? 

Thanks!
Chakra

inemp, 2021-05-06:
Yes I follow the steps :

```
$ git clone https://github.com/aws/aws-fpga  
$ cd aws-fpga
$ source sdk_setup.sh <-- crash
```

```
fpga_pci.c: At top level:
fpga_pci.c:455:1: error: conflicting types for ‘fpga_pci_get_address’
 fpga_pci_get_address(pci_bar_handle_t handle, uint64_t offset, size_t len,
 ^
In file included from fpga_pci_internal.h:20:0,
                 from fpga_pci.c:29:
../../include/fpga_pci.h:229:5: note: previous declaration of ‘fpga_pci_get_address’ was here
 int fpga_pci_get_address(pci_bar_handle_t handle, uint64_t offset
```

I just subscribed to the 'FPGA Developer AMI (Amazon Linux 2)' yesterday so 1.10.0 and "running on z1d.2xlarge".

AWScsaralay, 2021-05-06:
Hi, 

Unfortunately we are unable to reproduce this error when running on z1d.2xlarge, with Amazon Linux2. 

```
AMI Version:        1.10.0
Xilinx Version:     2020.2
Readme:             /home/ec2-user/src/README.md
AMI Release Notes:  /home/ec2-user/src/RELEASE_NOTES.md
GUI/Cluster setup:  https://github.com/aws/aws-fpga/blob/master/developer_resources
Developer Support:  https://github.com/aws/aws-fpga/blob/master/README.md#developer-support
```
```
[ec2-user@ip-172-31-57-124` aws-fpga]$ source sdk_setup.sh
INFO: Setting up environment variables
INFO: Changing AWS_FPGA_REPO_DIR from /home/ec2-user/src/project_data/aws-fpga to /home/ec2-user/aws-fpga
Entering /home/ec2-user/aws-fpga/sdk/userspace/utils
rm -f *.o *.a ../lib/libutils.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o io.o io.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o log.o log.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../include -I/usr/include   -c -o logger-kmsg.o logger-kmsg.c
mkdir -p ../lib
ar rcs ../lib/libutils.a io.o log.o logger-kmsg.o
Entering /home/ec2-user/aws-fpga/sdk/userspace/fpga_libs/fpga_pci
rm -f *.o ../../lib/libfpga_pci.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_pci.o fpga_pci.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_pci_sysfs.o fpga_pci_sysfs.c
mkdir -p ../../lib
ar rcs ../../lib/libfpga_pci.a fpga_pci.o fpga_pci_sysfs.o
Entering /home/ec2-user/aws-fpga/sdk/userspace/fpga_libs/fpga_dma
rm -f *.o ../../lib/libfpga_dma.a
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_dma_utils.o fpga_dma_utils.c
mkdir -p ../../lib
ar rcs ../../lib/libfpga_dma.a fpga_dma_utils.o
Entering /home/ec2-user/aws-fpga/sdk/userspace/fpga_libs/fpga_mgmt
rm -f *.o ../../lib/libfpga_mgmt.a ../../lib/so/libfpga_mgmt.so.1.0.0
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_mgmt_cmd.o fpga_mgmt_cmd.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_mgmt.o fpga_mgmt.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g -std=gnu99 -fPIC -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I. -I../../include -I/usr/include   -c -o fpga_hal_mbox.o fpga_hal_mbox.c
mkdir -p ../../lib
for library in ../../lib/libfpga_pci.a ../../lib/libutils.a ../../lib/libfpga_dma.a; do ar -x $library; done
ar rcs ../../lib/libfpga_mgmt.a *.o
mkdir -p ../../lib/so
cc -o ../../lib/so/libfpga_mgmt.so.1.0.0 -shared fpga_mgmt_cmd.o fpga_mgmt.o fpga_hal_mbox.o -Wl,-soname,libfpga_mgmt.so.1 -Wl,--whole-archive ../../lib/libfpga_pci.a ../../lib/libutils.a ../../lib/libfpga_dma.a -Wl,--no-whole-archive -lrt -lpthread
ln -sf libfpga_mgmt.so.1.0.0 ../../lib/so/libfpga_mgmt.so
Entering /home/ec2-user/aws-fpga/sdk/userspace/fpga_mgmt_tools/src
rm -f *.o *.a fpga-local-cmd static-fpga-local-cmd
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o fpga_local_cmd.o fpga_local_cmd.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o fpga_local_cmd_parse.o fpga_local_cmd_parse.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o virtual_jtag_pcie.o virtual_jtag_pcie.c
cc -DCONFIG_LOGLEVEL=0 -DFPGA_PCI_BARS_MAX=64 -DCLI_VERSION='"1.4.19"' -g  -Wall -Werror -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -I../../include/hal -I../../include -I../. -I.   -c -o virtual_jtag_server.o virtual_jtag_server.c
cc -o fpga-local-cmd fpga_local_cmd.o fpga_local_cmd_parse.o virtual_jtag_pcie.o virtual_jtag_server.o -L../../lib/so -lfpga_mgmt
cc -o static-fpga-local-cmd fpga_local_cmd.o fpga_local_cmd_parse.o virtual_jtag_pcie.o virtual_jtag_server.o -L../../lib -lfpga_mgmt -lrt -lpthread
Build complete.

Root privileges are required to install. You may be asked for your password...
Executing as root...

AWS FPGA: Copying Amazon FPGA Image (AFI) Management Tools to /usr/bin
[... cut ...]
AWS FPGA: Installing python bindings for FPGA management library
Installed /usr/lib64/python2.7/site-packages/fpga_dma.py
Installed /usr/lib64/python2.7/site-packages/fpga_mgmt.py
Installed /usr/lib64/python2.7/site-packages/fpga_pci.py
AWS FPGA: Done with Amazon FPGA Image (AFI) Management Tools install.
Done with SDK install.
INFO: sdk_setup.sh PASSED

```

Are you running this by launching a brand new instance with Amazon Linux2?
Does it work for you if launched on an instance with Centos FPGA Dev AMI?

Thanks!
Chakra

inemp, 2021-05-18:
Hello Chakra,

Indeed it worked. Works on Centos FPGA Dev AMI. Can be closed

closed 2021-06-22 by deeppat