aws/aws-fpga #530: No Platforms Found

wanyuzhan, 2021-07-29:
In AWS FPGA instance, when I want to create an application project in Vitis 2021.1. I cannot select a platform from repository. The platform I selected was xilinx_aws_vu9p-f1_shell-v04261818_201920_2.xpfm. The error shows in the doc. Shall you help me with this case?
[platform error.docx](https://github.com/aws/aws-fpga/files/6903581/platform.error.docx)

deeppat, 2021-07-29:
Hello @wanyuzhan, have you run vitis_setup.sh?

wanyuzhan, 2021-07-29:
Yes. I have. I followed this tutorial step-by-step.

deeppat, 2021-07-29:
Hi @wanyuzhan, Can you link to the tutorial so we can reproduce this issue?

wanyuzhan, 2021-07-30:
Hi Aws/Aws-Fpga,

Thanks for your help! I solve it by creating a new instance.

Wanyu

wanyuzhan, 2021-08-02:
Hi Deep,

Sorry to bother you, in order to run gzipc in Vitis GUI, the path is
Vitis_Libraries/data_compression/L2/tests/gzipc.
I tried to find all kernel codes in the gzipc project. Shall I know which
files belong to the kernel?
Also, some files were not found in the kernel when I did the software
emulation but they can be found in the host . Shall I know how to solve it?
[image: Screen Shot 2021-07-31 at 9 38 00 PM]
<https://user-images.githubusercontent.com/43864322/127759436-10f2bc7e-2efb-477b-98dd-620d9b893851.png>

Wanyu

deeppat, 2021-08-04:
@wanyuzhan Hi Wanyu, how are you running the emulation? Can you note down each step you took here so we can help further.

Thanks,

Deep

wanyuzhan, 2021-08-06:
Hi Deep,

Thanks for your reply. I changed to run gzip_app in Vitis GUI, the path is
Vitis_Libraries <https://github.com/Xilinx/Vitis_Libraries>/data_compression
<https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression>/L3
<https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L3>/
demos
<https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L3/demos>
/gzip_app/. In this case, I met some errors. I am expecting your reply and
help.
These are my following questions.

   - Shall I know which files belong to the kernel of the gzip_app project?
   - Some files were not found in the kernel when I did the software
   emulation but they can be found in the host.
   - [image: Screen Shot 2021-08-05 at 8.29.51 PM.png]
   - When I did kernel's software emulation, I encountered errors which are
   shown in this figure.
   - [image: Screen Shot 2021-08-05 at 8.32.15 PM.png]

These following steps show how I built and emulated the application
project.

   - I created a new application project in vitis 2021.1. AlsoI downloaded
   and installed xilinx_u250_gen3x16_xdma_3_1_202020_1 as the test platform.
   Are there any other plug-ins for U250 that I need to download ？
   - After I created the project, I clicked on the import sources inside
   the host. This figure shows all the codes I imported.
   - [image: Screen Shot 2021-08-05 at 8.30.29 PM.png]
   - Then I clicked on the import sources inside the kernel. This figure
   shows all the codes I imported.
   - [image: Screen Shot 2021-08-05 at 8.31.12 PM.png]
   - Then I created the kernel's binary container. In this case, I chose
   xilDecompress as the kernel.
   - After the kernel was created, I built the kernel's software emulation
   and got these errors.

Thanks for your help again. Looking forward to your response.

Wanyu

wanyuzhan, 2021-08-07:
Hi Aws/Aws-Fpga,

Thanks for your help, I have solved these problems.

Wanyu

deeppat, 2021-08-09:
Hi @wanyuzhan,

I see that you had installed the u250 platform files, however F1 is a different platform and you'd need to source vitis_setup.sh in order to download the platform. The environment variable: `$AWS_PLATFORM` should point to the platform file after vitis_setup.sh has run successfully! 

I am glad you were able to find a resolution. Please reopen this issue if you need more guidance!

Thanks,
Deep

closed 2021-08-09 by deeppat

wanyuzhan, 2021-08-09:
Hi Deep,

Thanks for your mention! Does it mean I can use the F1 platform in the
vitis_library project? If my guess is right, Where should I point the
environment variable?

Wanyu