aws/aws-fpga #532: Failed when run example: Vitis_Libraries/data_compression/L2/demos/gzip

wanyuzhan, 2021-08-20:
I tried to run gzip in Vitis 2021.1 GUI, the path is Vitis_Libraries/data_compression/L2/demos/gzip. The instance is xilinx_aws-vu9p-f1_shell-v04261818_201920_2.xpfm.In this case, I successfully compiled all codes. But I met some errors when I try to lunch SW_EMU, HW_EMU and hw. I am expecting your reply and help.
This is the error. 
[image]

deeppat, 2021-08-20:
Hi @wanyuzhan, can you list all steps that you ran to get to this error?
Were you able to test Vitis builds out in Non GUI mode? https://xilinx.github.io/Vitis_Accel_Examples/2021.1/html/compile_execute.html

wanyuzhan, 2021-08-20:
Hi Deep,

Thanks for your reply! These following steps show how I built and emulated the
application project.

   - I created a new application project in vitis 2021.1. Also I added
   xilinx_aws-vu9p-f1_shell-v04261818_201920_2.xpfm
   <https://github.com/aws/aws-fpga/blob/master/Vitis/aws_platform/xilinx_aws-vu9p-f1_shell-v04261818_201920_2/xilinx_aws-vu9p-f1_shell-v04261818_201920_2.xpfm>
as
   the test platform.
   - Then I clicked on the template which was vitis_library/L2/DEMOS/gzip
   to create the project.
   - Then I successfully compiled the software emulation. And I want to
   launch sw-emu. Then the error appeared as below.

[image: Screen Shot 2021-08-19 at 6.57.26 PM.png]

Then I try to run them on the command line. First I ran vadd example in
three emulations and got the results after launch. Then I tried to run
gzip. I use these following steps to launch emulation.
git clone https://github.com/aws/aws-fpga.git $AWS_FPGA_REPO_DIR
cd $AWS_FPGA_REPO_DIR
source vitis_setup.sh
cd gzip
make clean
make run TARGET=sw_emu DEVICE=$AWS_PLATFORM all

This is the result for sw-emu.

This is the result for hw-emu.
This is the result for hw.
Actually I didn't know why it prints “passed”. Shall you explain it to me?
Also, Do these results mean I succeeded? However, when I want to check
results in Vitis Analyzer, it tells me open_trace.csv and
opencl_summary.csv do not exist, so there’s no data in vitis analyzer.
These are all my problems. Looking forward to hearing from you!

Wanyu

wanyuzhan, 2021-08-20:
Hi Deep,

Thanks for your reply! These following steps show how I built and emulated the
application project.

   - I created a new application project in vitis 2021.1. Also I added
   xilinx_aws-vu9p-f1_shell-v04261818_201920_2.xpfm
   <https://github.com/aws/aws-fpga/blob/master/Vitis/aws_platform/xilinx_aws-vu9p-f1_shell-v04261818_201920_2/xilinx_aws-vu9p-f1_shell-v04261818_201920_2.xpfm>
as
   the test platform.
   - Then I clicked on the template which was vitis_library/L2/DEMOS/gzip
   to create the project.
   - Then I successfully compiled the software emulation. And I want to
   launch sw-emu. Then the error appeared as below.

[image]

Then I try to run them on the command line. First I ran vadd example in
three emulations and got the results after launch. Then I tried to run
gzip. I use these following steps to launch emulation.
git clone https://github.com/aws/aws-fpga.git $AWS_FPGA_REPO_DIR
cd $AWS_FPGA_REPO_DIR
source vitis_setup.sh
cd gzip
make clean
make run TARGET=sw_emu DEVICE=$AWS_PLATFORM all

This is the result for sw-emu.
[image]
[image]

This is the result for hw-emu.
[image]

This is the result for hw.
[image]

Actually I didn't know why it prints “passed”. Shall you explain it to me?
Also, Do these results mean I succeeded? However, when I want to check
results in Vitis Analyzer, it tells me open_trace.csv and
opencl_summary.csv do not exist, so there’s no data in vitis analyzer.
These are all my problems. Looking forward to hearing from you!

Wanyu

deeppat, 2021-08-30:
Hi @wanyuzhan , I think to get profile data, you need to enable it in the build or xrt.ini.

Can you check this for more info: https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/profilingapplication.html

deeppat, 2021-11-01:
Closing as the thread hasn't received a response in a while. Please re-open if you are still having any issues.

-Deep

closed 2021-11-01 by deeppat