aws/aws-fpga #537: Definition  FPGA maximum power limitation

iDoka, 2021-09-26:
Based on information on https://github.com/aws/aws-fpga/blob/master/hdk/docs/afi_power.md 

>> In order to help developers avoid these overpower events, the F1 system indicates a `afi-power-warning` on the CL interface (sh_cl_pwr_state[1:0]) when the FPGA power levels are above 85 watts, and the CL is in danger of having it's clocks disabled. This should allow the CL to self-throttle, or reduce power-hungry optimizations, and avoid having its input clocks disabled.
>> 
>> Power state of the FPGA `sh_cl_pwr_state[1:0]`:
>> * 0x0 – OK 
>> * 0x1 – UNUSED 
>> * 0x2 – afi-power-warning 
>> * 0x3 – afi-power-violation

If my understanding are correct the `afi-power-warning` event triggered when power consumption reached  85 watts.
Which power consumption will be triggered `afi-power-violation` event?

As a AWS F1 developer and customer I would to know **all usage limitations** of this type of instance.

AWScsaralay, 2021-09-29:
Hello,

The "**afi-power-warning**" is triggered when the AFI running on an F1 instance exceeds the power consumption limit of 85W.  The clocks from Shell to the CL are not yet gated while "**afi-power-warning**" is set. This is an indication that Customer Logic should reduce power consumption.

If the high power consumption is not mitigated then the Shell will assert "**afi-power-violation**" and gate the clocks to the CL. In addition, at the time of load, if "**afi-power-violation**" is set, clocks are disabled and any transactions along Shell-CL interface results in timeout.

For all practical purposes, we strongly recommend customer design to act on "**afi-power-warning**" to reduce power consumption.

Hope this helps. Please reach out to us if you have any questions. 

Thanks!
Chakra

deeppat, 2021-10-06:
Hi @iDoka, let us know if you need any more information and re-open this issue. I'll close it for now.