aws/aws-fpga #543: can't find xilinx_aws-vu9p-f1-04261818_dynamic_5_0.dsa file in aws_platform

umairqaisar000, 2021-12-20:
Please I can't find xilinx_aws-vu9p-f1-04261818_dynamic_5_0.dsa file in aws_platform, please upload it as I can;t load custom project on sdaccel.

AWScsaralay, 2021-12-20:
Hello,

Were you able to run example designs posted on github?
https://github.com/aws/aws-fpga/blob/master/SDAccel/examples/aws/helloworld_ocl_runtime/README.md#exclamation-please-note-xclbin--awsxclbin-file-formats-have-changed-from-sdx-20183-onwards-xclbin--awsxclbin-files-generated-using-earlier-sdx-versions-are-not-compatible-with-2018320191-based-xrts-if-you-are-using-a-2018320191-based-xrt-please-copy-over-awsxclbin--helloworld-executable-files-provided-in-the-20183_20191-subdirectory-to-this-folder

Could you also specif which tool version are you using? 

thanks!
Chakra

umairqaisar000, 2021-12-21:
I am using SDx 2018.3

AWScsaralay, 2021-12-21:
All the required files are downloaded during initial setup script. If not done already, could you please try running initial setup as described in the link below?
https://github.com/aws/aws-fpga/tree/master/SDAccel#github-and-environment-setup

umairqaisar000, 2021-12-21:
Okay Thank you will try and see if it works.

kyyalama2, 2022-04-12:
closing issue. Please feel free to reopen if required

closed 2022-04-12 by kyyalama2