aws/aws-fpga #546: AXI Model control parameters

abhaypelago, 2022-02-04:
I have the simulations with AXI models working. However, I see that model returns the read response and wr ack with fixed 1-2 clock latency. And all read responses are in-order.
Is there some control knobs in AXI model which can return these responses with random latency and out-of-order.
Also, I need control for various ready signals to be toggled randomly.
Thanks

czfpga, 2022-02-08:
Thank you for that information. I'll need to dig a bit into it to see if there is any built-in control knob in the testbench setup that can be directly utilized. I'll let you know asap. 

- Chen

AWScsaralay, 2022-02-25:
Hello,

All the Verification Modules, including the Shell BFM are provided [here](https://github.com/aws/aws-fpga/tree/master/hdk/common#verif)

The testbench does not provide parameters to vary the response time or behavior of AXI4. However, you could try modifying the behavior of the interfaces in [sh_bfm.sv](https://github.com/aws/aws-fpga/blob/master/hdk/common/verif/models/sh_bfm/sh_bfm.sv).

For example:
* [sh_bfm.sv#Line1099](https://github.com/aws/aws-fpga/blob/master/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L1099) shows condition when ```sh_cl_pcim_awready``` is set. 
* [sh_bfm.sv#Line1224](https://github.com/aws/aws-fpga/blob/master/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L1224) shows when ```sh_cl_pcim_rvalid``` is assert by the testbench.
You could tweak the behavior as per your requirement. 

However, please note that BFM changes you make does not conflict with the [Shell Interface Spec](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#ShellInterfaces). Otherwise your design behavior could differ between simulations and on real F1 instance. 

Please let us know if you have any questions or need anything. 

Thanks!
Chakra

AWScsaralay, 2022-03-14:
Hello, we'll mark this post as resolved. Please reach out to us if you have any questions. Thank you.

closed 2022-03-14 by AWScsaralay