aws/aws-fpga #549: How to use DSP Resources for dcmp operator?

RiddleSingh, 2022-02-16:
I am looking to find ways to implement a double-precision comparison block using DSP resources. By default the compiler makes use of "dcmp_64ns_64ns_1_2_no_dsp_1". [https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-bind_op](url) shows how we can change the resource for several other operators other than dcmp. 

In my application, I would like to implement 100 of this dcmp in parallel which is by default being implemented using LUT resources. This, however, leads to routing congestion and overutilization of LUT resources. I have plenty of DSP resources and I would like to instead make use of these. Can you suggest a way to do this?

Thanks

AWScsaralay, 2022-02-25:
Hello, 

We apologize for delayed response. We will reach out to Xilinx to understand if the resource sharing on DSP is a viable option.

Other option would be developing RTL design by specifically instantiating DSP primitives for required functions. Although this flow is going to be totally different than the current Vitis flow you are using. 

regards
Chakra

AWScsaralay, 2022-02-25:
Hello,

What we learned from Xilinx is --

Vitis does not provide a way for users to direct DCMP implementation using DSP48E2 primitives. The DSP block (56 bit) is not large enough to compare 64 bit double, so it would take two DSP blocks plus some number of LUTs to do the comparison. This would add cycles of latency. This is not a normally useful tradeoff for somebody doing floating point math, so it's not supported. It might be desirable for a customer only doing double compares and no other math, but that is not a normal use case. Also, if the customer cares about NAN and distinguishing between +/- 0, it would take 4 DSP blocks.

Please let us know if you have any questions.
Thanks!
Chakra

closed 2022-03-14 by AWScsaralay