aws/aws-fpga #551: Unable to simulate hello world HLx in vivado

t2shashwat, 2022-03-12:
Hello,

I am trying to simulate hello world program in vivado, located at _https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world_hlx_ . The simulation of this program **fails** at the **elaborate step**. It throws the below error and exits the simulation.

```
ERROR: [USF-XSim-62] 'elaborate' step failed with error(s) while executing '/home/shrivast/aws-fpga/hdk/cl/examples/cl_hello_world_hlx/example_projects/cl_hello_world.sim/sim_1/behav/xsim/elaborate.sh' script. Please check that the file has the correct 'read/write/execute' permissions and the Tcl console output for any other possible errors or warnings.

ERROR: [Vivado 12-4473] Detected error while running simulation. Please correct the issue and retry this operation.

ERROR: [Common 17-39] 'launch_simulation' failed due to earlier errors.

```
When I dig into the tcl console output, I find that a **dpi.so** file is not created inside project's xsim directory, which is also highlighted in the tcl console with the following message: 
```
Invalid path for DPI library: /home/shrivast/aws fpga/hdk/cl/examples/cl_hello_world_hlx/example_projects/cl_hello_world.sim/sim_1/behav/xsim/dpi.so
```
When I look further into tcl console, I find that **dpi_xsim.tcl** file was **not executed**. And it prints the following error in the console:

```
ERROR: [XSIM 43-3985] Problem with usage of options: unrecognised option '--additional_option'. Please see the usage below.

ERROR: Script failed:"/home/shrivast/aws-fpga/hdk/common/shell_v04261818/hlx/verif/scripts/dpi_xsim.tcl
```

**Steps to reproduce:** 
Follow below subheadings in this [link](https://github.com/aws/aws-fpga/blob/master/hdk/cl/examples/cl_hello_world_hlx/README.md) -

- Setup HLx Environment

- Create Example Design (GUI)

- Simulation (Code should throw errors here!)

Any help is appreciated.

Thanks,
Shashwat

kyyalama2, 2022-04-06:
Hi Shashwat

Sorry for the delayed response. Just wanted to check if you still need help with this and are seeing the issue or if this was resolved

Thanks

kyyalama2, 2022-04-12:
closing the issue

closed 2022-04-12 by kyyalama2