aws/aws-fpga #552: Cannot run hello world on AWS F1 FPGA

hecmay, 2022-03-19:
Hi,

I used the scripts in aws-fpga to set up environment and compile the hello world program from Vitis official repo on F1 instance. However, I was not able to run the registered `awsxclbin` binary with the host program. Here is the error I got

```shell
# Error from running "./host kernel.awsxclbin"
HOST: ip-172-31-87-64.ec2.internal
EXE: /home/centos/src/project_data/gaussian/host
[XRT] ERROR: See dmesg log for details. err=-
[XRT] ERROR: failed to load xclbin: Input/output error
Failed to program device[0] with xclbin file!
Failed to program any device found, exit!****

# Log from desmg
[ 5755.281977] xocl 0000:00:1d.0: icap.u.23068672 ffff8e123dc93410 __icap_peer_xclbin_download: peer xclbin download err: -5
[ 5755.291418] xocl 0000:00:1d.0: icap.u.23068672 ffff8e123dc93410 icap_cache_bitstream_axlf_section: found kind 6(MEM_TOPOLOGY)
[ 5755.291425] xocl 0000:00:1d.0: icap.u.23068672 ffff8e123dc93410 icap_cache_bitstream_axlf_section: found kind 26(ASK_GROUP_TOPOLOGY)
[ 5755.291432] xocl 0000:00:1d.0: icap.u.23068672 ffff8e123dc93410 icap_download_bitstream_axlf: err: -5
[ 5755.291437] xocl 0000:00:1d.0:  ffff8e12455d7098 xocl_read_axlf_helper: Failed to download xclbin, err: -5
[ 5755.308285] xocl 0000:00:1d.0:  ffff8e12455d7098 xocl_destroy_client: client exits pid(17594)
[ 5755.308291] xocl 0000:00:1d.0:  ffff8e12455d7098 xocl_drvinst_close: CLOSE 2
[ 5755.308295] xocl 0000:00:1d.0:  ffff8e12455d7098 xocl_drvinst_close: NOTIFY ffff8e123dc94c10
```

AWScsaralay, 2022-03-21:
Hello,

Thanks for reaching out. We would like to confirm couple of things with your setup:
1. Are you using the official Vitits github https://github.com/aws/aws-fpga/tree/master/Vitis?
2. Did you run `source vitis_setup.sh` before compiling?
3. Did you run `source vitis_runtime_setup` before running the example? 

Thank you,
Chakra

hecmay, 2022-03-21:
> Hello,
>
> Thanks for reaching out. We would like to confirm couple of things with your setup:
>
> 1. Are you using the official Vitits github https://github.com/aws/aws-fpga/tree/master/Vitis?
> 2. Did you [...]

@AWScsaralay yes for all. I used the Vitis examples based on the recommended commits. I also tried different FPGA developer image versions. 

Always got the same error.

kyyalama2, 2022-03-22:
Hi, Can you please confirm the instance type you are using and the dev AMI version you tried.

hecmay, 2022-04-06:
> Hi, Just checking if you still need help with this. Thanks

hi @kyyalama2. i was using the latest AMI. 

but we have switched to another environment. so it should be fine. thanks for your help

closed 2022-04-06 by hecmay