aws/aws-fpga #554: [Question]: DDR Burst Read

sajjadahmed677, 2022-04-20:
Hi everyone!
I am trying to simulate my design with DDR-C and found that it only supports fixed burst size (```ar_size and aw_size```) of value ```6```. which means it always do burst size of ```512``` bits. So,
my question is do other DDR controllers (```A, B and D```) also have same configuration or do they support variable burst size?

Another thing is how can i use this fixed configuration DDR with my 64bit design where i need multi beat bursts of 64 bit?

jacobmgn, 2022-04-20:
Hello,

Thanks for reaching out. 
We're looking into this and hope to have an answer for you soon.

sajjadahmed677, 2022-04-20:
Thanks @jacobmgn!
Looking forward to your response.

jacobmgn, 2022-05-02:
> I am trying to simulate my design with DDR-C and found that it only supports fixed burst size (ar_size and aw_size) of value 6. which means it always do burst size of 512 bits
> So, my question is do [...]

The simulation BFM ( https://github.com/aws/aws-fpga/blob/master/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L1510 ) has a fixed size of 6 for simplicity. This constraint may not match the actual shell implementation. You can explore different sizes by modifying the test bench/

- All of the DDR controllers should support variable burst size

You should be able to modify sh_bfm to take the actual ar_size and aw_size, but you will need to ensure compliance to the AXI specification.

As a side note, PCIM has a [limitation](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#pcim-interface----axi-4-for-outbound-pcie-transactions-cl-is-master-shell-is-slave-512-bit), though this doesn't apply for sh_DDR.

> Another thing is how can i use this fixed configuration DDR with my 64bit design where i need multi beat bursts of 64 bit?

- It is possible to implement narrow transfers by following the AXI protocol. There is another option, which is to use Xilin'x AXI data width converter 
  - https://docs.xilinx.com/v/u/en-US/pg059-axi-interconnect