aws/aws-fpga #555: Some image URLs are broken in documentation

whentojump, 2022-04-21:
Hi there,

Thank you for your work.

I notice there are several broken image links, like [here](https://github.com/aws/aws-fpga/blob/364e3733e3f93405b0d381d8a8dc1513d355b9cf/hdk/docs/RTL_Simulating_CL_Designs.md?plain=1#L18) and [here](https://github.com/aws/aws-fpga/blob/364e3733e3f93405b0d381d8a8dc1513d355b9cf/hdk/docs/RTL_Simulating_CL_Designs.md?plain=1#L295).

Through some preliminary `blame`ing, these screenshots were found available at least until 52b2a3a, and were somehow removed later.

Not a big problem, though :)

jacobmgn, 2022-04-25:
Thanks for reporting this, we're aware of the issue and it's been assigned

kyyalama2, 2022-05-18:
Dear customer,

Thank you so much for raising the issue and the relevant pointers. We are taking a look and will address the broken links.

Thanks