aws/aws-fpga #556: AXI "bvalid" behavior and latency

abhaypelago, 2022-05-01:
1. When is BVALID returned? 
a. Is it after write to DDR has been sent? 
b, Is it after write to PCIe bus from FPGA or ack from host?

2. What is the latency from WLAST assertion to BVALID assertion on both DDR and PCIM interfaces? If it's variable, what is the range?

3. For PCIM interface
a. After receiving BVALID for write to address Z, are subsequent reads to Z guaranteed to respond with previous write to Z.
b. If SH-PCIM accepts write to address A followed by write to address B, will the host memory be written in the same order?
c. If the answer to above question is NO, then, if CL waits for BVALID for address A and then writes to address B, will the host memory now guaranteed to write A and then write B.

czfpga, 2022-05-03:
Hi useragdp1,

I would recommend you run the [cl_dram_dma example simulation](https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_dram_dma/verif#readme) and check out the waveforms of the particular interface you're interested. In general, we don't specify this type of latency as it has too many dependencies on traffic, memory state, targeted address space, etc,.

For the ordering questions on PCIM, I would recommend you refer to AMBA AXI protocol for the write and read ordering. The PCIM is compliant to the AXI4 protocol with [a few limitations](https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Shell_Interface_Specification.md#pcim-interface----axi-4-for-outbound-pcie-transactions-cl-is-master-shell-is-slave-512-bit). 

Thanks,

- Chen

czfpga, 2022-05-10:
Hi useragdp1,

Is there any other questions you have? If not, I'll go ahead resolve this issue. Please feel free to open a new one for any new questions you might have in future. Thanks.

- Chen

kyyalama2, 2022-05-24:
Assuming resolved. Please feel free to re-open if you need further help