aws/aws-fpga #557: Vivado Fails to Synthesize Anything Other Than the Example Projects Using the Build Scripts

llee454, 2022-05-03:
# Problem:

I started an EC2 instance running the FPGA Developer AMI, and tried to use Vivado to create a new project that wraps custom Verilog Designs for use in Amazon F1 FPGAs. So far, I cannot get Vivado to synthesize anything other than the Demo projects.

# Steps:

   1.  On an EC2 instance running the FPGA Developer AMI: open a terminal, create a directory named "test_fpga", cd into the directory
    2. run source ~/.../aws-fpga/hdk_setup.sh
    3. run vivado
    4. When Vivado starts, click "Create a new Project". Select all defaults
    5. In the Tcl console run: aws::make_ipi
    6. This creates a block diagram in which the f1_inst block is created and not connected to anything. click "Run Synthesis".

Vivado returns an error message saying that it cannot find modules such as Microblaze. **The synthesis fails.**

# Conclusion:

Is there **any** documentation explaining how to create a new project and synthesize it?

I can synthesize the cl_hello_world example by running aws::make_ipi -examples cl_hello_world in the Tcl console, but I'm not asking how to customize one of these projects, I'm asking how to start a new Vivado project, add my own Verilog files, and connect those designs to the F1_inst block.

Please help, Thanks

AWScsaralay, 2022-05-03:
hello, 

For IPI designs please follow the instructions below:
1. https://github.com/aws/aws-fpga/tree/master/hdk#install-the-hdk-and-setup-environment
2. https://github.com/aws/aws-fpga/blob/master/hdk/docs/IPI_GUI_Vivado_Setup.md#linux-install
3. Please try this tutorial on creating an IPI design:
https://github.com/aws/aws-fpga/blob/master/hdk/docs/IPI_GUI_Examples.md#tutorial-on-how-to-create-hlx-ipi-cl_hello_world-example-using-ip-integration-flow

Please reach out to us if you have any questions. 

Thanks!
Chakra

llee454, 2022-05-03:
Thanks. I've tried using these resources however. When I run the steps outlined in (GUI Workflow with Vivado IP Integrator Quickstart Examples)[https://github.com/aws/aws-fpga/blob/master/hdk/docs/IPI_GUI_Examples.md#gui-workflow-with-vivado-ip-integrator-quickstart-examples], Vivado throws an error message once I click "Run Synthesis".

The instructions do not work.

AWScsaralay, 2022-05-05:
Hello,

Unfortunately we were unable to reproduce the error you are seeing. Following instructions seem to be working in our environment:

1. Launch F1.16xl Instance with FPGA Developer AMI | vivado tool version = 2021.2 | HDK Version = 1.4.23
2. Set up [IPI_GUI_Vivado_Setup](https://github.com/aws/aws-fpga/blob/master/hdk/docs/IPI_GUI_Vivado_Setup.md#linux-install)
3. https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world_hlx#setup-hlx-environment
```    
    $ git clone https://github.com/aws/aws-fpga.git $AWS_FPGA_REPO_DIR
    $ cd $AWS_FPGA_REPO_DIR
    $ source hdk_setup.sh
    $ mkdir -p ~/.Xilinx/Vivado
    $ echo 'source $::env(HDK_SHELL_DIR)/hlx/hlx_setup.tcl' >> ~/.Xilinx/Vivado/Vivado_init.tcl
```
4. [tutorial-on-how-to-create-hlx-design-example-from-scratch-using-ip--rtl-flow](https://github.com/aws/aws-fpga/blob/master/hdk/docs/IPI_GUI_Examples.md#tutorial-on-how-to-create-hlx-design-example-from-scratch-using-ip--rtl-flow)
5. ```aws::make_rtl```
6. Run Synthesis, successfully generated CL.post_synth.dcp

NOTE: The above steps only adds design template with **sh_ddr.sv** as top level. So my synthesized DCP only had sh_ddr.sv module. Please make sure to add the required RTL pieces to complete top level CL design as shown in the examples. 

Thanks!
Chakra

llee454, 2022-05-12:
Thank you for your prompt replies. I retested everything using and was able to get Vivado to compile my design. I noticed several issues that may help others.

kyyalama2, 2022-05-17:
Dear customer

Thanks for the feedback on prepare_new_cl and the cl_template.sv.
The prepare_new_cl.sh script is intended to only set up the directory structure thats similar to our cl_examples for customers to add their designs. The cl_template.sv itself is not a standalone design and is just intended as a place holder for customer design. We recommend to run one of our examples like cl_hello_world or cl_dram_dma for a quick start and add the design to cl_template when want to create custom_design and update the build scripts with corresponding source files. Please feel free to let us know if you have any follow up questions.

Thanks

kyyalama2, 2022-05-24:
Dear customer,

Just checking to make sure if there are any other issues that you need help with related to the ticket or if we can close the ticket.

Thanks

kyyalama2, 2022-05-31:
closing the ticket. please feel free to re-open or file a new ticket if you have any additional questions. Thanks

closed 2022-05-31 by kyyalama2