aws/aws-fpga #558: Notify scripts gives boto3.exception

jahagirdar, 2022-05-08:
```
wait_for_afi.py  --afi afi-<id> --notify --email <me>@<domain>.com
Traceback (most recent call last):
  File "/home/centos/src/project_data/aws-fpga/shared/bin/scripts/wait_for_afi.py", line 20, in <module>
    import boto3
  File "/home/centos/.local/lib/python2.7/site-packages/boto3/__init__.py", line 16, in <module>
    from boto3.compat import _warn_deprecated_python
  File "/home/centos/.local/lib/python2.7/site-packages/boto3/compat.py", line 20, in <module>
    from boto3.exceptions import PythonDeprecationWarning
  File "/home/centos/.local/lib/python2.7/site-packages/boto3/exceptions.py", line 43
    f"The '{service_name}' resource does not an API version of: {bad_api_version}\n"
                                                                                   ^
SyntaxError: invalid syntax
```
Similar issue was reported at https://github.com/aws-solutions/aws-instance-scheduler/issues/305 The solution there was to upgrade to python3 in aws-fpga scripts python2.7 is hardcoded in shebang.

kyyalama2, 2022-05-19:
Dear customer,

Thank you so much for reporting the broken feature to AWS. We are currently looking into this and will update once we have a fix. 

Thanks

kyyalama2, 2022-06-09:
Dear customer

Thank you so much for raising the issue. Also updated the wait_for_afi documentation to guide the customers on boto3 installation as mentioned above. Please feel free to re-open the issue if you need further help with this or still see a blocker issue with using wait_for_afi. Thank you so much for your interest in aws

jahagirdar, 2022-06-18:
After downgrading boto3 the script now gives the following error
AWS FPGA: (14:28:13) - Calling notification script to send e-mail to ...@......com
ImportError: No module named site

kyyalama2, 2022-06-20:
can you confirm the version of python you are using, with python --version. Thanks

kyyalama2, 2022-11-04:
closing this thread due to inactivity. If you still need help with this issue, please comment to re-open the thread. Thanks

closed 2022-11-04 by kyyalama2