aws/aws-fpga #559: prepare_new_cl.sh generates broken scripts and templates.

jahagirdar, 2022-05-08:
The script prepare_new_cl.sh in aws-fpga/hdk/cl/developer_designs/ generates broken code and scripts.
Issues noticed so far:

1. Code is generated as `design/cl_template.sv` While scripts in `build/scripts` references `cl_hello_world` everywhere.
2. Script references files which are not generated e.g `design/cl_id_defines.vh` and `common/design/cl_common_defines.vh`
3. The `design/cl_template.sv` throws 40+ errors during compilation. e.g.

```
ERROR: [Synth 8-2715] syntax error near . [/home/centos/src/project_data/aws-fpga/hdk/cl/developer_designs/vijay/build/src_post_encryption/cl_template.sv:183]
ERROR: [Synth 8-1031] cl_sh_pcis_awready is not declared [/home/centos/src/project_data/aws-fpga/hdk/cl/developer_designs/vijay/build/src_post_encryption/cl_template.sv:218]
ERROR: [Synth 8-1031] cl_sh_pcis_wready is not declared [/home/centos/src/project_data/aws-fpga/hdk/cl/developer_designs/vijay/build/src_post_encryption/cl_template.sv:220]
ERROR: [Synth 8-1751] cannot index into non-array tie_zero_id [/home/centos/src/project_data/aws-fpga/hdk/cl/developer_designs/vijay/build/src_post_encryption/cl_template.sv:291]
ERROR: [Synth 8-1751] cannot index into non-array tie_zero_id [/home/centos/src/project_data/aws-fpga/hdk/cl/developer_designs/vijay/build/src_post_encryption/cl_template.sv:292]

10 Infos, 1 Warnings, 1 Critical Warnings and 44 Errors encountered.
synth_design failed

```

jacobmgn, 2022-05-16:
Hello,

Thanks for reporting this issue, we're taking a look at it

Thanks!

kyyalama2, 2022-05-17:
Dear customer,

Thanks for reporting the issue. The prepare_new_cl.sh script is intended to only set up the directory structure thats similar to our cl_examples for customers to add their designs. The cl_template.sv itself is not a standalone design and is just intended as a place holder for customer design. We recommend to run one of our examples like cl_hello_world or cl_dram_dma for a quick start and add the design to cl_template when want to create custom_design and update the build scripts with corresponding source files. Please feel free to let us know if you have any follow up questions.
https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_hello_world
https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples/cl_dram_dma

Thanks

jahagirdar, 2022-05-18:
The minimum expectation from a template is that it should be clean start point for a new design. The current script is obliviously not a clean start-point. I ended up using a cleaned up template and folder layout from another site for my design.

In case Amazon AWS does not have the resources to fix this issue will they accept a PR?

closed 2025-04-21 by TrahStevAmz

jahagirdar, 2025-04-22:
Please provide closing comments. Is the script fixed, will not fix....

TrahStevAmz, 2025-04-23:
Hello,

The issue you described has been addressed in the F2 branch's [CL_TEMPLATE example](https://github.com/aws/aws-fpga/tree/f2/hdk/cl/examples/CL_TEMPLATE). It is now a self contained example that can be built and simulated.