aws/aws-fpga #565: [Quesstion]: Simulation of virtual jtag fcility

sajjadahmed677, 2022-05-28:
Hi!
I want to test my design in simulation which has an on-chip debug module with jtag interface. and i haven't found any example or way to test the jtag communication using virtual jtag and openocd in simulation. please guide me if there is possibility to test such system in aws-fpga system.

thanks,

czfpga, 2022-05-31:
Hi,

My understanding is that you have on on-chip debugger in your CL design and want to know if this debugger can be simulated. Is this a XIlinx IP, for example, ILA? Or this is your own design connected to the FPGA's external JTAG pins or virtual JTAG?

If it's a Xilinx core, like ILA, usually simulation is not support. Please refer to [PG172](https://docs.xilinx.com/v/u/en-US/pg261-system-ila) at p24 as an example. Can you please provide more information about your debugger?  

Thanks,

Chen

czfpga, 2022-06-02:
Hi @sajjadahmed677,

I think this might require behavior simulation model for some of the internal FPGA primitives, for example, for JTAG/vitural JTAG. I'm working on getting that information for you. One missing piece to me is that how do you plan to use the`openocd` to interact with the simulation? It sounds to me the simulation you're looking for is a RTL behavior simulation on your debugger.  Do you also expect to have a channel opened under the simulation environment to communicate with `openocd`? Or that's out of the scope of this issue. Can you please help clarify that? 

Thanks,

Chen

sajjadahmed677, 2022-06-02:
Hi @czfpga,

Yes i need a channel open for communication with ```openocd``` like a ```tcp/ip``` port. for that i would need to modify the the ```openocd``` driver for sure. 
One thing related to this discussion how can i use ```openocd``` to communicate with the virtual jtag on actual FPGA i have tried with default ```xvc``` driver available in ```openocd``` but it didn't work. so if you could direct me to any example which uses ```openocd``` for virtual jtag in ```aws-fpga``` environment. 

Thanks,

Sajjad

czfpga, 2022-06-06:
Hi @sajjadahmed677,

Sorry for the waiting. I found this from OpenOCD (https://openocd.org/doc/html/Debug-Adapter-Configuration.html#Debug-Adapter-Configuration). This might be the one you're already following, if not, might worth to check. 

Regarding to the simulation, I image your debugger logic would need to interact with the [CL_DEBUG ](https://github.com/aws/aws-fpga/blob/4750aacb4dac9d464b099b27e4337220cf0b0713/hdk/docs/Virtual_JTAG_XVC.md#embedding-debug-cores-in-the-cl). RTL behavior simulation should be supported, including the Xilinx debug bridge wrapped in that component. Please refer to https://github.com/aws/aws-fpga/blob/master/hdk/docs/RTL_Simulating_CL_Designs.md#rtl-simulation-for-verilogvhdl-custom-logic-design-with-aws-hdk for details. However, we currently don't support HW/SW co-simulation with OpenOCD. That could possibly be a very useful feature. I'll definitely bring this information back to the development team. Please continue following our repo to get announcements for new features we introduce in future. 

Thanks,

Chen

czfpga, 2022-06-10:
Hi @sajjadahmed677,

I want to check if all of your questions are answered. If so, I will go ahead mark this as resolved. Thanks,

Chen

czfpga, 2022-06-13:
Mark as resolved. Please feel free to open a new issue if there is any follow-up or new questions. Thanks.

Chen

closed 2022-06-13 by czfpga