aws/aws-fpga #570: XOCL installation failed.

Shahzaib2028, 2022-06-15:
I am getting error while installing XOCL on Vitis. I am following this link.
https://github.com/aws/aws-fpga/blob/master/Vitis/docs/XRT_installation_instructions.md

For 2021.1 Version, I used ```xrt_202110.2.11.0_7.9.2009-x86_64-aws.rpm``` rpm.

On running this  command ```scl enable devtoolset-6 bash```, it gives an error that this file ```devtoolset-6``` is not available in ```/etc/scl/conf/``` this folder, then I checked the folder and ```devtoolset-9``` file was available then I  replaced this command with ```scl enable devtoolset-9 bash``` ,  and it worked

When I run the build script ``` ./build.sh ``` , I got this error error. Please help me to understand this error and suggest me a solution to this.

```Built target msd
[ 68%] Built target mpd
make[2]: *** No rule to make target '/home/centos/src/project_data/aws-fpga/sdk/userspace/lib/libfpga_mgmt.a', needed by 'runtime_src/core/pcie/tools/cloud-daemon/aws/libaws_mpd_plugin.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2524: runtime_src/core/pcie/tools/cloud-daemon/aws/CMakeFiles/aws_mpd_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

OS : CentOS Linux release 7.9.2009 (Core)
Kernel version: 3.10.0-1160.49.1.el7.x86_64
Vivado version: 2021.1
Vitis: 2021.1

jacobmgn, 2022-06-16:
Hello,

Are you noticing this issue when using a normal CentOS installation or the [AWS Developer AMI](https://github.com/aws/aws-fpga/#fpga-developer-ami)?

kyyalama2, 2022-06-28:
Dear customer

Just following up to see if you can provide some details/confirmation on the AMI/OS used when you encountered this issue

Thanks

closed 2022-07-30 by Shahzaib2028