aws/aws-fpga #571: Unable to run FPGA dev AMI on m6a or m6i instances

kenta2, 2022-06-19:
M6 are the next generation after the M5 family of EC2 instances, with improved processors.  I tried m6a.xlarge, m6a.2xlarge, m6i.xlarge, and m6i.2xlarge and they all failed with the following error "The instance configuration for this AWS Marketplace product is not supported. Please see the AWS Marketplace site for more information about supported instance types, regions, and operating systems."  m5a (older hardware) do work.

Can you please enable the M6 family?

jelicicm, 2022-06-20:
Hi. I'm not employed with/related to AWS in any form, but I believe I can give a reply. M type instances do not feature FPGA boards in them. Only type of instance that does is F1: [https://aws.amazon.com/ec2/instance-types/f1/](url)
With M series, you get AMD EPYC CPUs and no external hardware.

kenta2, 2022-06-20:
On the M instances, I am only interesting in building (synthesizing) designs, using the Xilinx software that comes preloaded on the AMI.  This works for the M5 (and other families, like their suggested z1d family) but not M6.

I am not (yet) interested in running of FPGA; when I do become so, I'll copy my synthesized design over to the more expensive F1 instance via S3.

closed 2022-06-28 by kyyalama2

kenta2, 2022-06-28:
I confirm success for m6i, but I am also interested in the R6* family (more memory).  Can you please enable those as well?

You will probably have other customers interested in the C6 family as well.

kenta2, 2022-06-28:
I am unable to reopen this ticket, so will create a new ticket.