aws/aws-fpga #575: Does HLX flow still work?

Quarky93, 2022-07-05:
I'm confused if the HLX flow still works with Vivado 2021.2. After fixing #573 by commenting out the set_property is_locked line, the design successfully compiles and generates the TAR file.
However when generating the AFI, an error is encountered when running platform_verify.
This happens for all the HLX examples and my custom design.
Is there any way around this? Has anyone tried the HLX flow in the last 6 months?

czfpga, 2022-07-05:
@Quarky93,

Thank you for reporting this. Can you please share any building log syou might have that shows the platform_verify error? We could start from there. 

Thanks,

Chen

Quarky93, 2022-07-05:
Hi @czfpga , thanks for responding quickly. Here is the log from AFI generation, error is towards the bottom.

```
#-----------------------------------------------------------
# Vivado v2021.2 (64-bit)
# SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
# IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
# Start of session at: Mon Jul  4 05:10:29 2022
# Process ID: 1658
# Current directory: /home/builder/scripts
# Command line: vivado -mode batch -source ingest.tcl
# Log file: /home/builder/scripts/vivado.log
# Journal file: /home/builder/scripts/vivado.jou
# Running On: ip-10-0-130-150.ec2.internal, OS: Linux, CPU Frequency: 2499.998 MHz, CPU Physical cores: 4, Host memory: 66548 MB
#-----------------------------------------------------------
source ingest.tcl
# set userDCP "../checkpoints/SH_CL_routed.dcp"
# set awsDCP  "../checkpoints/SH_CL_BB_routed.dcp"
# set awsDCPStripped  "../checkpoints/SH_CL_BB_stripped.dcp"
# set powerDefaultRPT "../reports/power_report.default.rpt"
# set powerStaticRPT  "../reports/power_report.static.rpt"
# set timingRPT       "../reports/SH_CL_final_timing_summary.rpt"
# set ioRPT           "../reports/report_io.rpt"
# set utilRPT         "../reports/report_utilization.rpt"
# set partialBIT      "../bitstreams/SH_CL_final_pblock_CL_partial.bit"
# set partialLTX      "../bitstreams/SH_CL_final_pblock_CL_partial.ltx"
# set CL_PATH WRAPPER_INST/CL
# puts "Ingest start time: \[[clock format [clock seconds] -format {%a %b %d %H:%M:%S %Y}]\]"
Ingest start time: [Mon Jul 04 05:13:34 2022]
# set_param hd.supportClockNetCrossDiffReconfigurablePartitions 1
# set_param hd.platformVerifyCachedRun false
# check_integrity $userDCP
INFO: [Vivado 12-5531] Integrity check successful: /home/builder/checkpoints/SH_CL_routed.dcp
# set_msg_config -severity INFO -suppress
# set_msg_config -severity STATUS -suppress
# set_msg_config -severity WARNING -suppress
CRITICAL WARNING: [Common 17-1355] You are suppressing all messages of type 'WARNING'. You may potentially disregard important DRC, CDC, and implementation messages that can negatively impact your design.  If this is not desired, please run 'reset_msg_config -suppress -severity {WARNING}' to undo this change.
# set_msg_config -id {Chipscope 16-3} -suppress
# set_msg_config -string {AXI_QUAD_SPI} -suppress
# puts "Opening user DCP $userDCP: \[[clock format [clock seconds] -format {%a %b %d %H:%M:%S %Y}]\]"
Opening user DCP ../checkpoints/SH_CL_routed.dcp: [Mon Jul 04 05:13:37 2022]
# open_checkpoint $userDCP
CRITICAL WARNING: [Board 49-67] The board_part definition was not found for xilinx.com:f1_cl:part0:1.0. This can happen sometimes when you use custom board part. You can resolve this issue by setting 'board.repoPaths' parameter, pointing to the location of custom board files. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command.
CRITICAL WARNING: [Board 49-67] The board_part definition was not found for xilinx.com:f1_cl:part0:1.0. This can happen sometimes when you use custom board part. You can resolve this issue by setting 'board.repoPaths' parameter, pointing to the location of custom board files. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command.
Netlist sorting complete. Time (s): cpu = 00:00:03 ; elapsed = 00:00:03 . Memory (MB): peak = 3763.961 ; gain = 5.992 ; free physical = 58189 ; free virtual = 60511
Read XDEF File: Time (s): cpu = 00:00:23 ; elapsed = 00:00:23 . Memory (MB): peak = 6158.027 ; gain = 453.773 ; free physical = 55928 ; free virtual = 58305
Finished XDEF File Restore: Time (s): cpu = 00:00:23 ; elapsed = 00:00:24 . Memory (MB): peak = 6158.027 ; gain = 453.773 ; free physical = 55927 ; free virtual = 58305
Netlist sorting complete. Time (s): cpu = 00:00:00.04 ; elapsed = 00:00:00.04 . Memory (MB): peak = 6158.027 ; gain = 0.000 ; free physical = 55933 ; free virtual = 58314
open_checkpoint: Time (s): cpu = 00:02:43 ; elapsed = 00:04:48 . Memory (MB): peak = 6158.027 ; gain = 3460.090 ; free physical = 55930 ; free virtual = 58311
# reset_drc_check [get_drc_checks *]
# set_property IS_ENABLED TRUE [get_drc_checks *]
# set_property SEVERITY {ERROR} [get_drc_checks LUTLP*]
# delete_waivers [get_waivers -type DRC]
# puts "Checking User DCP is fully routed: \[[clock format [clock seconds] -format {%a %b %d %H:%M:%S %Y}]\]"
Checking User DCP is fully routed: [Mon Jul 04 05:18:25 2022]
# if { [llength [get_nets -hierarchical -filter { ROUTE_STATUS == "UNROUTED" }]] != 0} {
#   puts "***ERROR*** DCP has unrouted nets, ingestion failed, exiting..."
#   exit 1
# }
get_nets: Time (s): cpu = 00:00:56 ; elapsed = 00:00:26 . Memory (MB): peak = 6531.488 ; gain = 373.461 ; free physical = 55692 ; free virtual = 58073
# puts "Running pr_verify on initial DCP vs user DCP: \[[clock format [clock seconds] -format {%a %b %d %H:%M:%S %Y}]\]"
Running pr_verify on initial DCP vs user DCP: [Mon Jul 04 05:18:52 2022]
# pr_verify -full_check -in_memory -additional $awsDCP
Netlist sorting complete. Time (s): cpu = 00:00:06 ; elapsed = 00:00:06 . Memory (MB): peak = 7391.445 ; gain = 33.996 ; free physical = 54647 ; free virtual = 57028
resize_pblock: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 8193.430 ; gain = 77.027 ; free physical = 53923 ; free virtual = 56306
Read XDEF File: Time (s): cpu = 00:00:37 ; elapsed = 00:00:37 . Memory (MB): peak = 9169.609 ; gain = 709.297 ; free physical = 53078 ; free virtual = 55461
[... cut ...]
platform_verify: Time (s): cpu = 00:08:06 ; elapsed = 00:03:05 . Memory (MB): peak = 12725.691 ; gain = 32.012 ; free physical = 53838 ; free virtual = 56424
ERROR: [Common 17-39] 'platform_verify' failed due to earlier errors.

    while executing
"platform_verify -reference $awsDCPStripped -in_memory -cell WRAPPER_INST/CL"
    invoked from within
"if {![regexp {.*2017.4*} [version]]} {
    # run platform_verify if tool is not 2017.4 version
    puts "Running Platform Verify..."
    # following l..."
    (file "ingest.tcl" line 165)
```

czfpga, 2022-07-05:
Hi @Quarky93,

Thank you so much for providing the detailed report. I'll bring this to the dev team and we'll need to review this inside. I'll keep you updated on what we find.  

Thanks,

Chen

kyyalama2, 2022-07-08:
Dear customer

We are able to reproduce the exact error on our end and are in the process of debugging to figure out the reason for the error

ERROR: [Vivado 12-5763] platform_verify failed bitfile generation!
platform_verify: Time (s): cpu = 00:08:14 ; elapsed = 00:02:57 . Memory (MB): peak = 12604.809 ; gain = 32.012 ; free physical = 53889 ; free virtual = 56459
ERROR: [Common 17-39] 'platform_verify' failed due to earlier errors.

    while executing
"platform_verify -reference $awsDCPStripped -in_memory -cell WRAPPER_INST/CL"
    invoked from within
"if {![regexp {.*2017.4*} [version]]} {
    # run platform_verify if tool is not 2017.4 version
    puts "Running Platform Verify..."
    # following l..."
    (file "ingest.tcl" line 165)

We will update you shortly. 

Thanks

Quarky93, 2022-07-31:
Any updates on this?

kyyalama2, 2022-08-01:
Dear customer, Thank you so much for your patience and thanks a lot for checking on this. We are working on a fix for this and are in the process of testing the fixes. If everything goes well with testing, we should be able to release the fix very soon. We will update you shortly on this. 

Thanks

kyyalama2, 2022-08-09:
Dear customer, the issue in the AFI creation is fixed now. Please let us know if you still see any issues with the HLX flow. The manual fix for the tcl script mentioned in the earlier thread will be updated on github shortly.

Thanks

Quarky93, 2022-08-09:
Great news! I'll try it soon.

kyyalama2, 2022-08-19:
Dear customer, just following up to see if you are able to run the HLX flow successfully and/or if you have any comments/issues related to this.

Thanks

Quarky93, 2022-09-10:
@kyyalama2 Issue is resolved! Thanks very much!

closed 2022-09-10 by Quarky93