aws/aws-fpga #577: What is xdma0_user mapped to?

Quarky93, 2022-07-17:
Hi, I was under the assumption that xdma0_user would be mapped to one of the AXI-Lite interfaces similar to the standard Xilinx IP. However after some testing, I'm left confused. Can someone let me know is xdma0_user mapped to the sda, ocl, or bar1 axi-lite buses? Or must I use the AWS API? I would prefer to use xdma0_user to keep compatibility with on-prem cards using XDMA.

kyyalama2, 2022-07-19:
Dear customer

Thank you for your interest in AWS. The AWS FPGA PCIe Memory map is available at 
https://github.com/aws/aws-fpga/blob/master/hdk/docs/AWS_Fpga_Pcie_Memory_Map.md#aws-fpga-pcie-memory-map

Please refer to the table here for information on mapping of PF/BAR address spaces. Please let us know if this is helpful or if you are looking for any additional information that can help with your debug

Thanks

Quarky93, 2022-07-21:
Hi, I'm aware of the BAR mapping. However I'm less certain on what the xdma driver actually maps to on the xdma0_user character device specifically.

Normally with on-prem FPGA with XDMA, the xdma0_user is mapped to the AXI-Lite bus. I'm not sure what it does on AWS F1.

Quarky93, 2022-08-01:
Ok, xdma0_user is mapped to OCL :) .
@kyyalama2 should this be documented?

closed 2022-08-01 by Quarky93