aws/aws-fpga #579: Xiling 2022.1

bjourne, 2022-08-19:
Is the latest version of the tools installed on FPGA Developer AMI instances. However, the `vitis_setup.sh` script does not support that version. It's pretty easy to modify the script to make it work though.

kyyalama2, 2022-08-19:
Dear customer

Thank you so much for your interest in AWS and our latest developer AMI offering. We are in the process of releasing updates to our developer kit for 2022.1 version of xilinx tools. This should help with using the developer kit with 1.13 version of our developer AMI. We will update the ticket with updates. Thank you so much for your interest

Thanks

kromancer, 2022-09-01:
It is a bit sad to realize that the "Hello World" for Vitis will not work.
Things get a bit sadder because you have to pay for an EC2 instance to run it in.

kyyalama2, 2022-09-01:
Dear customer,

Thank you for your interest in AWS. Can you please confirm which version of developer AMI and developer kit are you using to run Vitis Hello World. As mentioned above, currently version 2022.1 developer kit is pending release. But Hello World for Vitis should work with the previous versions of developer AMI including 2021.2. So please let us know if you are seeing any issues with previous versions of tools. If this is a different issue that you want to request support for, please feel free to open a new ticket/issue for your case. This developer kit is intended to support EC2 F1 instances. So it does require EC2 F1 instances for running the examples as the workflow and AFIs are custom for F1 instances. 

Thanks

xlz-jbleclere, 2022-09-05:
Hi, please any ETA for the v1.13 developer kit support ?

kyyalama2, 2022-09-05:
Dear customer

Thank you so much for your interest in AWS. We are currently working on testing the dev kit updates and are targetting to release the dev kit this week. Also just in case you are not looking for 2022.1 version of the tools, the current developer kit works with 1.12.1 and below versions of developer AMI. We will post the updates here as we release the developer kit updates for 1.13 version of dev AMI/2022.1 version of xilinx tools.

xlz-epimor, 2022-09-29:
Hi do you have an updated status about the release of the aws dev kit for 2022.1 xilinx tools version ?

kyyalama2, 2022-10-04:
Dear customer

Thank you so much for your patience. We released a branch REL_2022_1 on github at https://github.com/aws/aws-fpga/tree/REL_2022_1 for 2022.1 version of xilinx tools. This branch fully supports the F1 flow; however please note the following known issues (also mentioned in Release notes):
    cl_uram example is currently not supported with 2022.1 version
    When loading a vitis design on an instance running HDK AFI/design, instance needs to be rebooted before loading the vitis design
    non-root access is unstable

Please feel free to contact us if you have any further questions.

Thanks
Kishore

Quarky93, 2022-10-27:
@kyyalama2 We are now using the REL_2022_1 branch.
What do you mean by 'non-root access is unstable'?
When will REL_2022_1 be promoted to main branch?
When can we expect REL_2022_2 😆

kyyalama2, 2022-10-27:
Dear customer

non root access referes to : https://github.com/aws/aws-fpga/tree/master/sdk#using-fpga-as-non-root-user
This currently is not working reliably with 2022.1 version running on dev AMI 1.13 (centos).

We are currently waiting for xilinx's fix on cl_uram example issues and further investigation into the item#2 on vitis loading after HDK design. In addition, we are also working on root causing the non-root access issue and the fix for it. So the promotion of the branch to aws-fpga currently is dependent on the three items listed above. Other than these three items, everything else should be functional on this branch.

2022.2 version of the tool is released by xilinx last week and we are int the process of evaluating the tool for compatibility and stability with our developer kit and the developer flow. We currently do not have an ETA for 2022.2 release. Will keep posted once we are able to validate the tool with our developer environment and ready for release. Thank you so much for your patience

Thanks

xlz-epimor, 2022-11-15:
Hi,
When will the AWS FPGA Developer AMI 1.13 be updated with new version of XRT for 2022.1 Xilinx tools (2.13.479) ?
When will the AWS FPGA Developer AMI for 2022.2 Xilinx tools be published ?
Thanks

kyyalama2, 2022-11-15:
Dear customer,

Thanks for your interest in our FPGA developer AMI. We are currently working with AMD/Xilinx on providing updates to our customers. However if you can kindly provide the information below, this will help us with 2022.2 developer AMI release
1.) are there any specific features in 2022.2 that you are interested in?
2.) any features or reason why you are interested in updated XRT version versus the current version provided?
3.) Any other information that you can provide related to what you are looking for in 2022.2 version and/or XRT updates?

Thanks

xlz-epimor, 2022-12-06:
Hi,
We are not looking for specific features, but we are developing tools that must be compatible with all Xilinx software/hardware versions and AWS is one of our deployment platforms.

Is the install of the gcc 5 via devtool 7 is planned on centos AMI for further release to support XRT C++ native API ?

Thanks

kyyalama2, 2022-12-07:
Dear customer

Thank you so much for the additional information and your feedback. We currently do not have an ETA on the items requested. We have added the requested items to our list. We will provide updates once we have them available. Please feel free to provide us any additional information on the request you have/want to provide.

Thanks
Kishore

xlz-epimor, 2022-12-08:
Hi
Thanks for your answers. I still have one more request about AMI 1.13. I cannot find in Vitis/docs/XRT_installation_instructions.md xrt packages associated to Xilinx 2022.1 tools. Could you provide updated packages ?
Thanks,
Edouard

kyyalama2, 2022-12-08:
Hi Edouard

Thanks for your request. In the meantime, also want to point you to instructions we have at https://github.com/aws/aws-fpga/blob/REL_2022_1/Vitis/docs/XRT_installation_instructions.md#centosrhel-build-and-install-steps. Please note that these instructions are updated for 2022.1 version in this branch. These instructions can be used to build XRT rpms for Centos developer AMI. Please feel free to let us know if you have any questions related to these instructions.

Thanks
Kishore

kyyalama2, 2022-12-08:
Hi Edouard

Also updated pre-compiled rpms in the table. 

Thanks
Kishore

ljp0101, 2023-02-18:
Some related feedback: It would be really helpful if y'all published an Ubuntu FPGA Developer AMI. CentOS 7 is a painfully creaky experience with a modern cloud tech stack where all the software is EOLed within five years and there are issues with a lot of the still supported but older stuff easily resolved by upgrading to the latest LTS. The old version of OpenSSL in CentOS 7 in particular has caused me a lot of grief including capping Node at v16 and sometimes intractable build and other compatibility issues with a lot of things including AWS' own C++ SDK. And I'm stuck with the AMI when I need things that require the Vivado license.

@xlz-epimor FYI, you can run F1s on newer XRT if you want. I'm now running on XRT 2022.02 / Ubuntu 22.04 pulled from the a base dockerfile Xilinx publishes on its github.

kyyalama2, 2023-02-21:
Dear customer

Thank you so much for your feedback. We are evaluating various options including providing ubuntu AMIs due to Centos support issues. We will provide any relevant updates as we make progress. The details provided in your feedback are very helpful

Thanks

kyyalama2, 2023-06-01:
closing this ticket since no outstanding issue related to this ticket currently. Thank you so much for your feedback. This is very helpful. Please feel free to reopen if you have any follow up questions related to this ticket. Thank you so much for your continued interest in F1

Thanks

ljp0101, 2024-03-12:
VSCode dropped support.

https://code.visualstudio.com/docs/remote/faq#_can-i-run-vs-code-server-on-older-linux-distributions