aws/aws-fpga #581: [Question]: FPGA Peer2Peer Build Process

sajjadahmed677, 2022-09-23:
Hi developers,

I have a question regarding build process of the designs having Peer2Peer communication between FPGAs. How would we build the design which will be ported to different FPGAs? e.g, 
I have a design which has two partitions ```A and B.``` 
The Partition ```A``` will be ported to slot0 FPGA and ```B``` will be ported to slot1 FPGA.
Do i have to build these two parts individually? or is there a way to build one AFI and that could be ported to two FPGAs by FPGA management tools. Like can i specify in build process that certain module will be ported to different FPGA or its separate builds for each FPGA.

Your help is highly appreciated.

Thanks,

AWScsaralay, 2022-09-23:
Hello, 

Peer2Peer is a hardware level functionality supported in F1 between two FPGA cards. However, an AFI corresponds to a single image that will be loaded into an FPGA slot. Therefore in your case, you may need to run two different builds for ```A``` and ```B``` designs and hence create two different AFIs to ultimately load into two different FPGA slots on an F1 instance. 

Please contact us if you have any questions. 

Thanks!
Chakra

AWScsaralay, 2022-09-26:
Hello, 
Unfortunately the simulation testbench provided by AWS is limited to using a single Shell BFM which works with a single CL design. There is no active support for Peer2Peer simulations in the existing testbench. 

Please note that [sh_bfm.sv](https://github.com/aws/aws-fpga/blob/master/hdk/common/verif/models/sh_bfm/sh_bfm.sv#L857) is provided as an unencrypted source allowing users to potentially experiment with the testbench. 

Thanks!
Chakra

kyyalama2, 2022-11-04:
Dear customer, closing this thread since there were no follow up questions. Please feel free to comment if you need further help and we will re-open the thread. Thanks

closed 2022-11-04 by kyyalama2