aws/aws-fpga #583: ERROR: [BD 5-336] This command cannot be run, as the BD-design is locked.

svpolonsky, 2022-10-20:
Hi,

I can't generate a bitstream for Vitis' hello_world example. I would appreciate your advice on how I can get the bitstream generated.

I get the following error message:
```
ERROR: [BD 5-336] This command cannot be run, as the BD-design is locked. Locked reason(s):
* Block design contains locked IPs. Please run report_ip_status for more details and recommendations on how to fix this issue.
List of locked IPs:
cl_clk_main_a0_250Mhz_reset_0
cl_clk_extra_b0_250Mhz_reset_0
cl_clk_extra_c0_500Mhz_reset_0
cl_clk_extra_a1_125Mhz_reset_0
cl_HIP_0
cl_axi_gpio_null_0
cl_aws_0_0

```

I attach the log <aws-fpga>/Vitis/examples/xilinx_2021.2/hello_world/_x.hw.xilinx_aws-vu9p-f1_shell-v04261818_201920_3/link/vivado/vpl/runme.log

[runme.log](https://github.com/aws/aws-fpga/files/9832875/runme.log)

AWScsaralay, 2022-10-21:
Hello, 

Could you please confirm if you are trying to build hello_world example based on the instructions provided here?
https://github.com/aws/aws-fpga/tree/master/Vitis#github-and-environment-setup
https://github.com/aws/aws-fpga/tree/master/Vitis#build-the-host-application-and-xilinx-fpga-binary

Thanks!
Chakra

svpolonsky, 2022-10-22:
HI Chakra, I confirm I use these instructions

kyyalama2, 2022-10-26:
Dear customer,

can you please confirm if you are running this on an AWS provided FPGA developer AMI or if using on-premise development. Also can you please confirm if you are seeing this error at https://github.com/aws/aws-fpga/tree/master/Vitis#build-the-host-application-and-xilinx-fpga-binary step?

If using AWS FPGA developer AMI, can you please provide the AMI version you are using and if you are using the top of the tree developer kit or a different version of the dev kit.

We are unable to reproduce the error. On our end, we are able to successfully generate .xclbin for the hello world example.So any information on the environment you have setup for running the hello world example will be helpful. In addition, can you confirm if you ran any other flow on the cloned directory like HDK RTL flow/HDK IPI flow before running the vitis hello world.

Thanks

svpolonsky, 2022-10-28:
Hi Kyyalama2,
(1) I use FPGA Dev AMI
AMI Version:        1.12.1
Xilinx Version:     2021.2
(2) yes, I get this error at "Build the Host Application and Xilinx FPGA Binary" step:
`make TARGET=hw DEVICE=$AWS_PLATFORM all`
(3) I think I use the top of the tree developer kit - I downloaded it just few days ago.
Thanks!

kyyalama2, 2022-11-01:
Dear customer,

as per our live sync up and the email thread; can you please confirm the problem is now resolved with launching a new instance using the dev AMI and running through the vitis flow? Please feel free to let us know if you need any further help on this.

Thanks
Kishore

svpolonsky, 2022-11-01:
Hi Kishore, I do confirm the issue has been resolved. Thanks a lot!

closed 2022-11-01 by svpolonsky