aws/aws-fpga #584: EC2 F1: AWS FPGA DRAM DMA Example fails during execution

jasmisbvb, 2022-11-02:
We are currently evaluating AWS's EC2 F1 instances as solution for a multitude of FPGA projects. As part of that we are running the example code provided by AWS in [https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples ](https://github.com/aws/aws-fpga/tree/master/hdk/cl/examples).
Observed behaviour:
- The Hello World example builds and executes successfully.
- The DRAM DMA example builds successfully and executes without crashing, but the C code logic in the example returns with an error and fails:
"
ERROR, test_dram_dma.c +145: dma_example(): DIMM 0 failed with 16711498 bytes which differ.
"
Please refer to the attached screenshot for the complete output.
The error repeats for all 4 DIMMs. The number of differing bytes is different on each execution of the example code.

Expected behaviour:
- Just like the Hello World example, the DRAM DMA example finishes its execution successfully. It's comparison of data written to/read from memory through the C code does not find any differing bytes.

Performed steps for each of the examples:
1. Setup the HDK on an EC2 instance with AWS FPGA Developer AMI. Then build the FPGA CL of the example by following the steps described on [https://github.com/aws/aws-fpga/blob/master/hdk/README.md ](https://github.com/aws/aws-fpga/blob/master/hdk/README.md)
Result of step: AFI is indicated as available in AFI storage.
2. Setup the SDK on an EC2 F1 instance with AWS Linux Base AMI according to [https://github.com/aws/aws-fpga/blob/master/sdk/README.md ](https://github.com/aws/aws-fpga/blob/master/sdk/README.md) also tried using latest FPGA Developer AMI (Amazon Linux 2)
3. Build the XDMA driver and load it into the kernel according to [https://github.com/aws/aws-fpga/blob/master/sdk/linux_kernel_drivers/xdma/xdma_install.md ](https://github.com/aws/aws-fpga/blob/master/sdk/linux_kernel_drivers/xdma/xdma_install.md)
4. Load the AFI as described on the hdk/sdk READMEs.
5. Build the C code host application part of the example and then execute it as described on the same READMEs.
7. The observed behaviour occurs.

It is important for us that the DRAM DMA example works correctly, as performant memory access through XDMA is essential for our FPGA projects.

Please support here, is it possible that the DRAM DMA example or the XDMA driver are faulty?

kyyalama2, 2022-11-04:
Dear customer, Thanks for your interest in AWS F1. Just to make sure, can you confirm if you see Busmaster enabled when you do lspci. Thanks

rokil, 2022-11-07:
I am a colleague of @jasmisbvb. Here is the output, looks like busmaster is missing?

`# lspci`
`00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma] (rev 02)`
`00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]`
`00:01.1 IDE interface: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II]`
`00:01.3 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 01)`
`00:02.0 VGA compatible controller: Cirrus Logic GD 5446`
`00:03.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)`
`00:1c.0 Non-Volatile memory controller: Amazon.com, Inc. NVMe SSD Controller`
`00:1d.0 Memory controller: Amazon.com, Inc. Device 1042`
`00:1e.0 Memory controller: Amazon.com, Inc. Device 1041`
`00:1f.0 Unassigned class [ff80]: XenSource, Inc. Xen Platform Device (rev 01)`
lspci gives identical output on both F1 instances with:
- FPGA Developer AMI
- Base Linux AMI with AWS FPGA SDK installed as well as xmda loaded to the kernel (appearing in lsmod).

kyyalama2, 2022-11-07:
Dear customer, 

can you please post the output of lspci -vv for the  00:1d.0 Memory controller: Amazon.com, Inc. Device 1042 and 
00:1e.0 Memory controller: Amazon.com, Inc. Device 1041. This should show whether the Busmaster is enabled or not

Thanks

rokil, 2022-11-09:
For `00:1d.0 Memory controller: Amazon.com, Inc. Device 1042` it says `BusMaster+`
For `00:1e.0 Memory controller: Amazon.com, Inc. Device 1041` it says `BusMaster-`

<details>

`00:1d.0 Memory controller: Amazon.com, Inc. Device 1042`
`        Subsystem: Xilinx Corporation Device 0007`
`        Physical Slot: 29`
`        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-`
`        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-`
`        Latency: 0`
`        Region 0: Memory at 82000000 (32-bit, non-prefetchable) [size=32M]`
`        Region 1: Memory at 85400000 (32-bit, non-prefetchable) [size=2M]`
`        Region 2: Memory at 85600000 (64-bit, prefetchable) [size=64K]`
`        Region 4: Memory at 2000000000 (64-bit, prefetchable) [size=128G]`
`        Capabilities: [40] Power Management version 3`
`                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)`
`                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-`
`        Capabilities: [60] MSI-X: Enable+ Count=33 Masked-`
`                Vector table: BAR=2 offset=00008000`
`                PBA: BAR=2 offset=00008fe0`
`        Capabilities: [70] Express (v2) Endpoint, MSI 00`
`                DevCap: MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us`
`                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W`
`                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-`
`                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+`
`                        MaxPayload 128 bytes, MaxReadReq 512 bytes`
`                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-`
`                LnkCap: Port #0, Speed 8GT/s, Width x16, ASPM not supported, Exit Latency L0s unlimited, L1 unlimited`
`                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+`
`                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-`
`                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-`
`                LnkSta: Speed 8GT/s, Width x16, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-`
`                DevCap2: Completion Timeout: Range BC, TimeoutDis+, LTR-, OBFF Not Supported`
`                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled`
`                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+, EqualizationPhase1+`
`                         EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-`
`        Kernel driver in use: xocl`
`        Kernel modules: xocl`

`00:1e.0 Memory controller: Amazon.com, Inc. Device 1041`
`        Subsystem: Xilinx Corporation Device 0007`
`        Physical Slot: 30`
`        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-`
`        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-`
`        Region 0: Memory at 85618000 (64-bit, prefetchable) [size=16K]`
`        Region 2: Memory at 8561c000 (64-bit, prefetchable) [size=16K]`
`        Region 4: Memory at 85000000 (64-bit, prefetchable) [size=4M]`
`        Capabilities: [40] Power Management version 3`
`                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)`
`                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-`
`        Capabilities: [70] Express (v2) Endpoint, MSI 00`
`                DevCap: MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us`
`                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 25.000W`
`                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-`
`                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+`
`                        MaxPayload 128 bytes, MaxReadReq 512 bytes`
`                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-`
`                LnkCap: Port #0, Speed 8GT/s, Width x16, ASPM not supported, Exit Latency L0s unlimited, L1 unlimited`
`                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+`
`                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-`
`                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-`
`                LnkSta: Speed 8GT/s, Width x16, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-`
`                DevCap2: Completion Timeout: Range BC, TimeoutDis+, LTR-, OBFF Not Supported`
`                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled`
`                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-`
`                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-`

</details>

kyyalama2, 2022-11-15:
Dear customer,

can you please try out the following:

1.) enable busmaster and mem space on both the PFs using
setpci -s <bdf> 0x4.l=0x00000006
2.) Also do you see this failure on other instances too? If you havent can try launching another instance and see if you can load the same AFI and run the test successfully or if it still fails?
3.) In addition, just to make sure, did you make any changes to the RTL/cl example design? Did it pass timing?
4.) Do you see any hangs or protocol errors in the shell metrics
ref: https://github.com/aws/aws-fpga/blob/master/hdk/docs/HOWTO_detect_shell_timeout.md#how-to-detect-a-shell-timeout-has-occurred

Thanks

kyyalama2, 2023-01-11:
Dear customer,

Can you please confirm if you are still seeing this issue or the recommendations resolved the errors you were seeing?

Thanks

kyyalama2, 2023-02-22:
closing this ticket, since not sure if still require support on this. Please feel free to comment if need help on this. Thanks